clash-lang/clash-compiler #154: Modules are being unnecessarily recompiled

basile-henry, 2016-07-02:
For example I have two top entities in modules A and B, they both depend on module C. When I compile them module C is being compiled twice.

Would it be possible to reuse the C.hi and C.o files created when compiling the first time for the top entity in module A so it can be reused to compile the top entity in module B?

As far as I understand, only one top entity can be compiled at a time, so in this example it would go like this (assuming A.hs, B.hs and C.hs are in src/):

```zsh
clash --vhdl -isrc/ src/A.hs
clash --vhdl -isrc/ src/B.hs
```

When compiling A.hs, it also compiles C.hs which creates C.hi and C.o
In the second command, is it possible for me to force clash to use the generated C.hi and C.o in order to compile B.hs?

I also tried `clash --vhdl -isrc/ src/A.hs src/B.hs` and it behaves the same way (C is compiled twice).

christiaanb, 2016-07-02:
So what's going one here is that, for simulation (both the interpreted and compiled), we have GHC apply different optimisations than we do for synthesis. In the past, this would sometimes lead to weird errors which could be solved by either:
- Deleting the .hi and .o files
- Running with `-fforce-recomp`
To be new-user-friendly, we opted to do the latter by default whenever we do synthesis (and it can currently not be disabled).

So that is why you always see CLaSH re-compiling (from Haskell to GHC Core) the `C.hs` file, because we want to make sure that the right optimisations are applied to the function in `C.hs`.

Now, in principle we could re-use synthesised definitions from `C.hs` (or at least their monomorphic/first-order versions). This would mean serialising the intermediate representations in some way. To be completely forthright, I don't see myself investigating such serialisation anytime soon... I simply don't have the time right now...

martijnbastiaan, 2021-03-19:
This has been tackled (albeit in a slightly different way) by the [Clash starter projects](https://github.com/clash-lang/clash-starters). It loads definitions from interface files and allows you to pick op entities using `-main-is` (see [flags documentation](https://clash-lang.readthedocs.io/en/latest/developing-hardware/flags.html)), hence preventing recompilation.

closed 2021-03-19 by martijnbastiaan