clash-lang/clash-compiler #156: <no location info>

tomjaguarpaw, 2016-07-04:
I got the error

````
<no location info>:
    CLaSH Error:
CLaSH.Netlist(282): Not in normal form: Not a variable reference or primitive as subject of a case-statement
````

Would it be possible to add the location info?  It took me about twenty minutes to track down the source of the error.

martijnbastiaan, 2017-10-19:
I've struggled with this too, two workarounds exist:

 * Build with `stack build --profile`. Then, compile your program with `stack exec clash -- Example.hs -prof +RTS -xc`. Compilation is slow, but profiling information is super useful.
* Use `./clash-dev` to start an `clashi` instance. Use `HasCallStack` to add location info in the relevant places.

christiaanb, 2020-01-16:
We track the location these days, even more so when you compile with `-g`

closed 2020-01-16 by christiaanb