clash-lang/clash-compiler #162: Improving performance of Unsigned/Signed and Fixed

tomjaguarpaw, 2016-07-19:
Slow simulation is my single biggest pain point with CLaSH at the moment.

Could we do some work on improving the performance of Unsigned/Signed and Fixed?  The former is phenomenally (40x) slower than Integer, and the latter is 3x slower than the former.  If we could get down to Word64 speeds (four thousand times faster than Fixed) that would be absolutely fantastic.

Some sample code.  Here is `src/Profile2.hs`:

````haskell
{-# LANGUAGE DataKinds #-}
{-# LANGUAGE KindSignatures #-}
{-# LANGUAGE MagicHash #-}

import CLaSH.Prelude
import GHC.Word (Word64)
import System.Environment (getArgs)

seven :: Int
seven = 7

loop_Word64 :: Word64
loop_Word64 = go big 0
  where go n acc = if   n <= 0
                   then acc
                   else acc `seq` go (n - 1) (acc + n)
        big = 10 ^ seven

loop_Integer :: Integer
loop_Integer = go big 0
  where go n acc = if   n <= 0
                   then acc
                   else acc `seq` go (n - 1) (acc + n)
        big = 10 ^ seven

loop_Unsigned :: Unsigned 64
loop_Unsigned= go big 0
  where go n acc = if   n <= 0
                   then acc
                   else acc `seq` go (n - 1) (acc + n)
        big = 10 ^ seven

loop_Fixed :: Fixed Unsigned 64 0
loop_Fixed = go big 0
  where go n acc = if   n <= 0
                   then acc
                   else acc `seq` go (n - 1) (acc + n)
        big = 10 ^ seven

main = do
     arg <- fmap Prelude.head getArgs
     case arg of
          "Word64"   -> print loop_Word64
          "Integer"  -> print loop_Integer
          "Unsigned" -> print loop_Unsigned
          "Fixed"    -> print loop_Fixed
````

This is the output of running it

````
ghc -O2 -XDataKinds src/Profile2.hs && for i in Word64 Integer Unsigned
Fixed; do echo $i; time src/Profile2 $i; echo --; echo; done
Word64
50000005000000

real    0m0.004s
user    0m0.000s
sys     0m0.000s
--

Integer
50000005000000

real    0m0.156s
user    0m0.156s
sys     0m0.000s
--

Unsigned
50000005000000

real    0m6.364s
user    0m6.340s
sys     0m0.024s
--

Fixed
50000005000000.0

real    0m15.490s
user    0m15.460s
sys     0m0.036s
--
````

tomjaguarpaw, 2016-07-19:
Most of the slowdown relative to `Integer` comes from `fromInteger_INLINE`.

christiaanb, 2016-07-21:
I've started on improving simulation speed in the following branch of the clash-prelude: https://github.com/clash-lang/clash-prelude/tree/fast_sim