clash-lang/clash-compiler #163: Could not find module ‘GHC.TypeLits.Normalise’

beojan, 2016-07-27:
I get this message immediately after installing clash using stack (`stack install --resolver=lts-6.9 clash-ghc`).

Presumably similar to #137 but the fix given there doesn't work.

christiaanb, 2016-07-27:
Although it is a bit outdated, do some of the suggestions in: https://github.com/clash-lang/clash-compiler/wiki/FAQ#how-do-i-run-and-install-clash-using-stack work?

i.e. calling clash like so:

```
stack exec --resolver=lts-6.9 clash -- --interactive
```

beojan, 2016-07-27:
Yes, that works.

christiaanb, 2016-08-18:
@beojan Can this issue be closed?

beojan, 2016-08-18:
Yes, but I wonder *why* `stack install` doesn't work here?

christiaanb, 2016-08-18:
The problem is that, `clash` being a specialised `ghc`, it needs the package databases to find the installed packages (i.e. clash-prelude, and some type-checker plugins); however, `stack` doesn't install these packages in one of the "default" package databases  (the user, or global package database). Instead, `stack` installs the packages in a separate package database.

By calling `stack` with `--resolver=lts-6.9`, certain environment variables are set to which these separate package databases are added; consequently, `clash` is then able to find the installed packages.

ggreif, 2016-08-18:
@christiaanb here is a curious fact (bug in GHC?)
When I compile a multi-module program using `clash`, say `A`, `B`, where `B` depends on `A`. I the first compilation I see
``` log
[1 of 2] Compiling A     ( A.hs, A.o )
[2 of 2] Compiling B     ( B.hs, B.o )
```
which is okay. The second compilation, without touching any file, gives:
``` log
[1 of 2] Compiling A     ( A.hs, A.o ) [GHC.TypeLits.Normalise changed]
[2 of 2] Compiling B     ( B.hs, B.o ) [GHC.TypeLits.Normalise changed]
```
OTOH, when I `touch A.hs`, I get
``` log
[1 of 2] Compiling A     ( A.hs, A.o )
[2 of 2] Compiling B     ( B.hs, B.o ) [GHC.TypeLits.Normalise changed]
```
Can you reproduce this behaviour? I am using `stack` (`lts-6.11` if that matters), but have seen it also when using purely from the command line.

Clearly this is somewhat botherful, as some of my modules take significant time to compile.

christiaanb, 2016-08-18:
@ggreif: yes, see also https://github.com/clash-lang/ghc-typelits-natnormalise/issues/2

It might be a GHC issue... we don't know.

closed 2018-12-18 by christiaanb