clash-lang/clash-compiler #166: Lists with >64 bits values -> sign extension?

jgjverheij, 2016-08-07:
Hi,

I think there is a problem with CλaSH values having more than 64 bits.
Let's first show an example with Integers:
```
*Main> let a = 10000000000000000000
*Main> let b = a + 1
*Main> [a .. b]
[10000000000000000000,10000000000000000001]
```
Works as expected, but when I change the type of 'a' to 'Signed 68' the output is different:
```
*Main> let a = 10000000000000000000 :: Signed 68
*Main> let b = a + 1
*Main> let c = [a .. b]
*Main> a
10000000000000000000
*Main> b
10000000000000000001
*Main> c
[-8446744073709551616,-8446744073709551615]
*Main> pack a
0000_1000_1010_1100_0111_0010_0011_0000_0100_1000_1001_1110_1000_0000_0000_0000_0000
*Main> pack $ P.head c
1111_1000_1010_1100_0111_0010_0011_0000_0100_1000_1001_1110_1000_0000_0000_0000_0000
```
The first 4 (67-64) bits are '1' instead of '0'. This is also the case by using 'Unsigned' values. It seems like a sign-extension problem, because using a smaller value for 'a' does not give problems.

christiaanb, 2016-08-07:
I think the problem might be the `[a .. b]`, which is uses `enumFromTo` of the `Enum` class, which (erroneously) uses `Int` internally, which cannot represent values outside `-2^63 .. 2^63-1.

You can see the problem if you do: `fromEnum a`.

Anyhow, I'll fix this in the `clash-prelude` package.

christiaanb, 2016-08-18:
Fixed by: https://github.com/clash-lang/clash-prelude/commit/abe737f1ddcda28d637c3237b8d0d47b7e16232d

closed 2016-08-18 by christiaanb