clash-lang/clash-compiler #170: Explain why to use :> instead of Cons

tomjaguarpaw, 2016-08-19:
There is a warning to use :> instead of Cons.  Could you also explain in the warning the reason for it?

ggreif, 2016-08-19:
I guess `(:>)` is an abstract interface, `Cons` the concrete one. Possibly later the concrete interface won't be exposed to the module users.

As an aside, I have a little project underway to do run-length-encoded `Vec`tors for simulation, which would not have `Cons` any more.

tomjaguarpaw, 2016-08-19:
If that's the case it would be helpful to have "`Cons` will be deprecated in future" in the docs.

christiaanb, 2016-08-19:
I'll update the documentation.

What @ggreif said is correct, `:>` is the abstract interface, while `Cons` is the concrete one. 

The reason is a bit silly/sad: While the clash compiler happily supports the constructor aspect of GADTs, it (currently) does not support the pattern matching aspect (see the end of chapter 4 of my thesis if you want to know why). So I added the pattern synonym (`:>`), which is based on `head` and `tail` for the pattern-matching side, which are supported by the compiler.

Hopefully, if/when clash supports GADT patterns, the (`:>`) will become the concrete interface.

martijnbastiaan, 2019-01-31:
GADT support has recently landed in Clash (https://github.com/clash-lang/clash-compiler/pull/465). We *can* now use `Cons` :). We should remove the warning and remove the pattern synonyms currently present.

christiaanb, 2020-01-16:
The warning has been removed. We keep the pattern synonym for the reasons described in: https://github.com/clash-lang/clash-compiler/pull/966

closed 2020-01-16 by christiaanb