clash-lang/clash-compiler #171: BlockRam Docs, arguments swapped in example?

drhodes, 2016-08-21:
From [this page](http://hackage.haskell.org/package/clash-prelude-0.10.14/docs/CLaSH-Prelude-BlockRam.html#usingrams), in the first `cpu` example:

```
dataMem :: Signal MemAddr -- ^ Read address
        -> Signal MemAddr -- ^ Write address
        -> Signal Bool    -- ^ Write enable
        -> Signal Value   -- ^ data in
        -> Signal Value   -- ^ data out
dataMem wr rd en din = mealy dataMemT (replicate d32 0) (bundle (wr,rd,en,din))
  where
    dataMemT mem (wr,rd,en,din) = (mem',dout)
      where
        dout = mem !! rd
        mem' | en        = replace wr din mem
             | otherwise = mem
```

It seems to me that the first argument to dataMem should be `rd` instead of `wr`, is that right?  My guess is that since, in the next section of code that calls dataMem, the call swaps those args too, so it works as expected.

```
    memOut = dataMem wrAddr rdAddr wrEn aluOut
```

Am I missing something?

christiaanb, 2016-08-21:
No... you're not missing anything... The haddock comments accompanying the function are swapped. The first argument is the write address, and the second argument is the read address.

I'll fix the docs tomorrow.

drhodes, 2016-08-21:
Aha, that explains it. Thanks for the quick clarification.

closed 2016-08-22 by christiaanb