clash-lang/clash-compiler #174: Not in normal form: RHS of case-projection is not a variable:

tomjaguarpaw, 2016-09-06:
I get `Not in normal form: RHS of case-projection is not a variable:` when I try to compile this:

````haskell
{-# LANGUAGE DataKinds     #-}

module NoInlineExample where

import           CLaSH.Prelude

{-# ANN example
  (defTop { t_name = "example"
          , t_inputs = [ "a"
                       ]
          , t_outputs = [ "b"
                        , "c"
                        ]
          }
  )#-}

example :: Signal (BitVector 1) -> Signal (BitVector 1, BitVector 1)
example input = foo $ bundle (input, pure 0)

{-# NOINLINE foo #-}
foo :: Signal (BitVector 1, BitVector 1)
    -> Signal (BitVector 1, BitVector 1)
foo input = input
````

The full error is

````
$ clash --vhdl example.hs 
[1 of 1] Compiling NoInlineExample  ( example.hs, example.o )
Loading dependencies took 0.789128s
Applied 26 transformations
Normalisation took 0.187161s

example.hs:18:1:
    CLaSH.Netlist(220): Not in normal form: RHS of case-projection is not a variable:

case input50 of
  CLaSH.Signal.Internal.:-
    (ww1 :: GHC.Tuple.(,)
              (CLaSH.Sized.Internal.BitVector.BitVector 1)
              (CLaSH.Sized.Internal.BitVector.BitVector 1))
    (ww2 :: CLaSH.Signal.Internal.Signal'
              (CLaSH.Signal.Internal.Clk system 1000)
              (GHC.Tuple.(,)
                 (CLaSH.Sized.Internal.BitVector.BitVector 1)
                 (CLaSH.Sized.Internal.BitVector.BitVector 1))) ->
    GHC.Prim.(#,#)
    @(GHC.Tuple.(,)
        (CLaSH.Sized.Internal.BitVector.BitVector 1)
        (CLaSH.Sized.Internal.BitVector.BitVector 1))
    @(CLaSH.Signal.Internal.Signal'
        (CLaSH.Signal.Internal.Clk system 1000)
        (GHC.Tuple.(,)
           (CLaSH.Sized.Internal.BitVector.BitVector 1)
           (CLaSH.Sized.Internal.BitVector.BitVector 1)))
      ww1
      ww2
    
    NB: The source location of the error is not exact, only indicative, as it is acquired after optimisations.
    The actual location of the error can be in a function that is inlined.
    To prevent inlining of those functions, annotate them with a NOINLINE pragma.
````

If I replace `foo input = input` by `foo = id` then it compiles successfully.

tomjaguarpaw, 2016-09-06:
And if I remove the `NOINLINE` pragma then it compiles successfully.

tomjaguarpaw, 2016-09-06:
The same result with `-fno-specialise`:

````
$ clash --vhdl -fno-specialise example.hs                                                                 
[1 of 1] Compiling NoInlineExample  ( example.hs, example.o )
Loading dependencies took 0.807069s
Applied 26 transformations
Normalisation took 0.179213s

example.hs:18:1:
    CLaSH.Netlist(220): Not in normal form: RHS of case-projection is not a variable:

case input50 of
  CLaSH.Signal.Internal.:-
    (ww1 :: GHC.Tuple.(,)
              (CLaSH.Sized.Internal.BitVector.BitVector 1)
              (CLaSH.Sized.Internal.BitVector.BitVector 1))
    (ww2 :: CLaSH.Signal.Internal.Signal'
              (CLaSH.Signal.Internal.Clk system 1000)
              (GHC.Tuple.(,)
                 (CLaSH.Sized.Internal.BitVector.BitVector 1)
                 (CLaSH.Sized.Internal.BitVector.BitVector 1))) ->
    GHC.Prim.(#,#)
    @(GHC.Tuple.(,)
        (CLaSH.Sized.Internal.BitVector.BitVector 1)
        (CLaSH.Sized.Internal.BitVector.BitVector 1))
    @(CLaSH.Signal.Internal.Signal'
        (CLaSH.Signal.Internal.Clk system 1000)
        (GHC.Tuple.(,)
           (CLaSH.Sized.Internal.BitVector.BitVector 1)
           (CLaSH.Sized.Internal.BitVector.BitVector 1)))
      ww1
      ww2
    
    NB: The source location of the error is not exact, only indicative, as it is acquired after optimisations.
    The actual location of the error can be in a function that is inlined.
    To prevent inlining of those functions, annotate them with a NOINLINE pragma.

````

christiaanb, 2016-09-07:
The above commit fixes the bug in the `0.6` branch, this bug doesn't show in `master` due to alternative means to disable the worker-wrapper induced by CPR analysis in GHC 8.

commit 1419e014 (2016-09-07): Disable strictness analysis, again

Strictness analysis induces demand analysis, which induces CPR
analysis. Normally, CPR analysis can be disabled by `-fcpr-off`,
however, due to https://ghc.haskell.org/trac/ghc/ticket/10696,
CPR analysis cannot be completely disabled.

The problem is not CPR analysis itself, but the worker-wrapper
combo that follows as a result of the CPR analysis. See commit
721fcfa9198925661cd836668705f817bddaae3c for more details on
how this W/W-combo breaks CLaSH.

Fixes #174
closed 2016-09-07 by christiaanb