clash-lang/clash-compiler #181: Error about non-existent recursive cycles

adamwalker, 2016-11-06:
When compiling this file: 
https://github.com/adamwalker/clash-expts/blob/master/Cordic.hs

I get this error:
<no location info>:
    CLaSH error call:
    CLaSH.Normalize(182): Callgraph after normalisation contains following recursive cycles: [[Cordic.magnitude'_a463]]

It simulates fine and I cant spot any recursive cycles in the code.
The error happens with clash-ghc version 0.6.23. I would test it with the latest version but I cant seem to figure out how to get Stack to upgrade packages.

ggreif, 2016-11-06:
Regarding `stack`, just ask for the latest lts-6 resolver.

adamwalker, 2016-11-06:
I managed to upgrade to 0.6.24 and the issue still exists.

Thanks for the tip regarding the resolver. I set it with the "--resolver" command line argument to stack. Do you know how I can make this the default? Setting the resolver field in my global stack config (~/.stack/config.yaml) does not have the desired effect.

christiaanb, 2016-11-06:
Figured out what's going on here:

Sometimes (still haven't figured out exactly when), GHC turns a recursive let-binding/where-clause into a recursive function which looks like:

```haskell
f x = .... (f x) ...
```

i.e. a function that recursively calls itself with the same argument, AND, the argument is unused except for those recursive calls. Anyhow, CLaSH has a transformation that removes this recursive call and turn everything into a recursive let-binding again. This transformation is part of the normalisation process.

What happened in your case is that normalisation didn't happen, because the `consts` top-level binding is a list of `Double` that needs to be converted to `Fixed` at compile-time by CLaSH. But because `Double` is considered non-representable by CLaSH it refuses to normalise functions that result in a `Double`. If you run clash with `-clash-debug DebugFinal`, you'll see:

```
CLaSH.Normalize(149): Expr belonging to bndr: Cordic.magnitude'_ws13 (:: Cordic.Complex
  (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 10 10)
-> CLaSH.Sized.Vector.Vec
     4
     (Cordic.CordicState
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 10 10)
        GHC.Types.Double)) has a non-representable return type. Not normalising:
```

Anyhow, as a work-around, you'd have to figure out how to make the `consts` top-level binding to be of type `Vec 4 (SFixed 10 10)`. As a fix, I should make `Float` and `Double` "representable" and ensure that CLaSH at leasts support the to and from `Rational` conversions.

christiaanb, 2016-11-06:
Another problem why this particular issue shows up, is that `last` is a primitive (used in `ifoldl`: http://hackage.haskell.org/package/clash-prelude-0.10.14/docs/src/CLaSH-Sized-Vector.html#ifoldl), and CLaSH currently has no specialised logic to "unroll" its definition. Consequently, it did not inline the `magnitude'_ws13` binding, which would've solved the problem as well. So yeah, I have to "bugs" to fix :-)

closed 2016-11-07 by christiaanb

christiaanb, 2016-11-07:
The fix is also backported to the 0.6 branch. You'll get some ominously sounding warnings about "might not normalise" (unless you raise the inlining limit as suggested in the warning text), however, CLaSH will successfully create VHDL/Verilog.

christiaanb, 2016-11-07:
Actually, it seems that on the `0.6` branch, you need to run clash with `-clash-inline-limit=115` or you hit a bug in the ANF transformation. I don't know whether I'll fix that bug... because I'd rather focus my efforts on `master`.

adamwalker, 2016-11-07:
Thanks for the workaround and quick fix! Just curious - what's the ETA on 0.7?

christiaanb, 2016-11-08:
So the "proper" work-around, is to not let the code default to `Double` by being explicit about the return type of `CordicState`; for example by saying:

```haskell
magnitude' :: Complex (SFixed 10 10) -> SFixed 10 10
magnitude' c = realPart (cplx $ (cordic $ CordicState c 0 :: CordicState (SFixed 10 10) (SFixed 10 10)))
```

With regards to releasing 0.7, hopefully by the end of this month.

adamwalker, 2016-11-08:
Oh, I hadn't realised that I had not constrained the second type parameter of CordicState. I guess this is not really a Clash bug, just a confusing error message. Sorry for the noise.

commit 41c72226 (2016-11-07): Reduce non-representable `last` and `init`

Fixes #181
commit 6c8fe1f5 (2016-11-07): Reduce non-representable `last` and `init`

Fixes #181