clash-lang/clash-compiler #182: Clash compiler generates wrong systemverilog (missing type)

idzardh, 2016-11-09:
I tested all three HDL's and only the systemverilog compiler outputs code that gives an error in QuestaSim. The error generated by questasim is as follows:

`** Error: ./Testbench_v1_types.sv (954): Failed to find name 'array_of_3_logic_vector_1_to_lv' in specified scope`

It also generates the same error for two more locations. It appears as though the whole declaration of this signal is missing, which is strange as the error is not reported in VHDL or Verilog. 

This is were the error occurs:

 `function array_of_3_array_of_3_logic_vector_1 array_of_3_array_of_3_logic_vector_1_cons(Testbench_v1_types::array_of_3_logic_vector_1 x,logic [0:2][0:0] xs [0:1]);
    array_of_3_array_of_3_logic_vector_1_cons[0] = Testbench_v1_types::array_of_3_logic_vector_1_to_lv(x);
    array_of_3_array_of_3_logic_vector_1_cons[1:2] = xs;
  endfunction`

idzardh, 2016-11-10:
Based on some further research it seems like the error is in the v2bv function, which would also make sense since the conversion (in this instance) is from a vector of three bits to a bitvector of length three.

The function is used as follows: 

`(a,b,c) = foldl g (0, 0, False) $ zipWith3 f conf1 conf2 (repeat sig)
  where
    g x y = if snd y then y else x
    f conf1p (conf2a, conf2b)p sig = h $ zipWith3 i conf1p conf2b (repeat sig)
      where
        i conf1pp (conf2ba, conf2bb) sig = conf1pp && conf2ba <= sig && conf2bb >= sig
        h vb = (conf2b, (v2bv . map bool2bit) vb, reduceOr vb`

This function triggered the error as described above. However changing the `reduceOr` to `foldr1 (||)` reduced one error. 

By the way the basic idea is to set a number of parameters based on an address (that's why there is the double comparison).

christiaanb, 2016-11-17:
The error is not in the `v2bv` function specifically. What happened is that it seems that CLaSH wasn't didn't "collect" all the types to put in the `types.sv` file. Could you give me some code that actually produces the incorrect SystemVerilog? I'm unable to even typecheck the above code.

idzardh, 2016-11-18:
I have made a more elaborate example that displays the bug, the code is as follows:

```haskell
module Test where

import CLaSH.Prelude

dummyMealyT :: (Vec 4 ( Vec 3 Bool ), Vec 4 ( Vec 3 (BitVector 22, BitVector 22), Unsigned 4 ))
            -> Bit -> BitVector 32 -> (Bit, (Unsigned 4, BitVector 3, BitVector 32))

dummyMealyT (conf1, conf2) st addrIn = (resp, outp)
  where
    dest = slice d31 d10 addrIn
    (outp, resp) = ((rId,rFrag, addrIn), rSel)

    (rId, rFrag, rSel) = foldl g (0, 000, 0) $ zipWith3 f conf1 conf2 (repeat dest)
      where
        g x y = if bit2bool (trd y) then y else x

        f conf1p (conf2a, conf2b) dest = addId $ zipWith3 comp conf1p conf2a (repeat dest)
          where
            comp conf1pp (conf2aa, conf2ab) dest = bool2bit $ conf1pp && conf2aa <= dest && conf2ab >= dest
            addId vB = (conf2b, v2bv vB, reduceOr vB)

conf1 = replicate d2 ( False :> True :> False :> Nil ) ++ replicate d2 ( True :> True :> False :> Nil )

conf2 = zip conf3 conf4

conf3 = map truncAddr $ a :> b :> c :> d :> Nil
  where
    truncAddr = map (\(x,y) -> (truncateLSB x, truncateLSB y))
    truncateLSB vec = slice d31 d10 (fromIntegral vec :: BitVector 32)
    a = ((0,2047) :> (2048,4095) :> (4096,6143) :> Nil)
    b = ((6144,8191) :> (8192,10239) :> (10240,12287) :> Nil)
    c = ((12288,14335) :> (14336,16383) :> (16384,18431) :> Nil)
    d = ((18432,20479) :> (20480,22527) :> (22528,24575) :> Nil)

conf4 = (1::Unsigned 4) :> 2 :> 3 :> 4 :> Nil

trd (a,b,c) = c

bool2bit bool = if bool then 1 else 0

bit2bool bit = 1==bit

topEntity = mealy (dummyMealyT (conf1, conf2)) (0::Bit)
```

The generated SV is compiled using QuestaSim 10.4c (64 bits). The first file that needs to be compiled is the types file, but it gives the same error as mentioned before. (This time on line 124)

christiaanb, 2016-11-18:
OK, thanks for the test case, I'll try to fix it over the weekend

christiaanb, 2017-04-10:
I cannot duplicate the behaviour for this bug in the current `master` branch. With which version of clash did you test this?

idzardh, 2017-04-10:
This was generated with clash 0.6.23, I haven't tested it since

christiaanb, 2017-04-10:
No longer happens in `master` -> closing.

closed 2017-04-10 by christiaanb