clash-lang/clash-compiler #184: SystemVerilog: zip does not generate the same code as writing the vector by hand

idzardh, 2016-11-14:
A range of addresses is generated as follows: 

``` haskell
testAddresses = singleton (
  (truncateLSB 0, truncateLSB (2^11-1) ) :>
  (truncateLSB (2^11), truncateLSB (2^13-1) ) :>
  (truncateLSB (2^13), truncateLSB (2^30-1)) :> Nil )
  ++ singleton (
  (truncateLSB 2^30, truncateLSB (2^30+2^15-1) ) :>
  (truncateLSB (2^30+2^15), truncateLSB (2^30+2^22-1) ) :>
  (truncateLSB (2^30+2^22), truncateLSB (2^31-1)) :> Nil )

truncateLSB vec = slice d31 d10 (fromIntegral vec :: BitVector 32)
```

And if a `zip` is applied with the vector `1 :> 2 :> Nil` this generates incorrect SystemVerilog (but correct VHDL). The compilation does not return any errors but the first two address tuple are 0 and only the final tuple is correctly set. So the output would look something like this:

```systemverilog
{ ( (0,0), (0,0), (8192,       1073741824) )
, ( (0,0), (0,0), (1077936128, 2147493647) )
}
``` 

However the following code:

``` haskell
testAddresses = singleton ( (
  (truncateLSB 0, truncateLSB (2^11-1) ) :>
  (truncateLSB (2^11), truncateLSB (2^13-1) ) :>
  (truncateLSB (2^13), truncateLSB (2^30-1)) :> Nil ), 1)
  ++ singleton ( (
  (truncateLSB 2^30, truncateLSB (2^30+2^15-1) ) :>
  (truncateLSB (2^30+2^15), truncateLSB (2^30+2^22-1) ) :>
  (truncateLSB (2^30+2^22), truncateLSB (2^31-1)) :> Nil ), 2)
```

Does generate correctly functioning SystemVerilog.

christiaanb, 2017-04-10:
This will be fixed by: https://github.com/clash-lang/clash-compiler/pull/213/commits/a38e1a51b1e27c45a80d0b5909f879904a1d4587

christiaanb, 2017-04-10:
Fixed by https://github.com/clash-lang/clash-compiler/commit/fe4077a8b7a21592799018f3584bf3d886fc3b77

closed 2017-04-10 by christiaanb