clash-lang/clash-compiler #191: Parse error with proc

LambdaScientist, 2016-12-01:
I noticed this when trying to make a function called "proc"

It causes a parse error.

I was able to reproduce it in CLaSH interactive by doing info on "proc"

CLaSH.Prelude> :info proc
<interactive>:1:1: parse error on input ‘proc’

You do not get the error when you slightly rename "proc"

CLaSH.Prelude> :info proc2
<interactive>:1:1: Not in scope: ‘proc2’

LambdaScientist, 2016-12-01:
I should note this is solved by simply renaming away from "proc"

ggreif, 2016-12-01:
@LambdaScientist 

`proc` is a reserved word https://wiki.haskell.org/Keywords#proc

`-XNoArrows` may unreserve it.

LambdaScientist, 2016-12-01:
I thought it was related to that. GHCI did not react in the same way. 

```
PS P:\HaskellCode\ClashByExample> ghci
GHCi, version 7.10.3: http://www.haskell.org/ghc/  :? for help

Prelude> :info proc
<interactive>:1:1: Not in scope: `proc'
Prelude>

```

On GHCI "instance" gives the parse error but "qualified" gives the scope. 

On CLaSH "instance" gives the parse error but "qualified" gives the scope. 

Why does CLaSH give a different output for "proc" but the same for "instance" and "qualified"?

christiaanb, 2016-12-01:
For historical reasons, `-XArrows` was/is enabled by default in CLaSH. In the next version it will not be however.

closed 2016-12-01 by LambdaScientist