clash-lang/clash-compiler #193: Still got this problem with Ghc_7_2 and  7_3. It wouldnt even compile basic  code

Chalolennox, 2016-12-22:
module First where

import CLaSH.Prelude

counter = s
  where
    s = register 0 (s + 1)
topEntity = counter

ggreif, 2016-12-22:
You mean GHC v7.10.3? It should compile with `:load First.hs`, but it won't with `:vhdl`. Add a type signature to `topEntity` to instruct the compiler about the desired bit width.

Chalolennox, 2016-12-22:
yes. thx for the fast reply

Chalolennox, 2016-12-22:
@ggreif, I hope you can bare with me for a while, cause I got tons of question to this incredible tool

ggreif, 2016-12-23:
Sure, just go and ask!

Chalolennox, 2016-12-28:
@ggreif , here is a simple code
 how would i use iterate or generate with this function?

module _AND where
import CLaSH.Prelude as CP
import qualified Prelude

_and::Signal Bool->Signal Bool->Signal Bool
_and a b  = ( a .&&. b )

topEntity =_and

Chalolennox, 2016-12-28:
@ggreif  to create a circuit with four AND-gates connected to each other

Chalolennox, 2016-12-28:
@ggreif ,sry for the dumb question. I have solved it  using fold
[image]

Chalolennox, 2016-12-28:
I would like to keep this open for any beginner like me. I hope it's possible.

Chalolennox, 2016-12-30:
@christiaanb one last question. I have managed to compile a mealy state machine with Quartus Prime Lite Edition and I was wondering whether the vhdl code is optimized  so taht you can view the circuit with the state machine viewer?

christiaanb, 2017-01-02:
@Chalolennox The VHDL is not generated in such a way that the quartus software can infer a state machine diagram from it. The circuit will still be efficient however.

Chalolennox, 2017-01-03:
@christiaanb Thx so much for your responce. I have managed to compile a mealy state machine and its quite what I expected. I am really a big fun of your work guys and hope  that clash soon becomes a standard. I will keep this open for some weeks, cause am still experimenting. thx

Chalolennox, 2017-03-13:
One last question before I close this: It is possible to import other libraries, say like yampa, to experiment with FRP concepts? obvious  problems are of course limitations  mentioned in the  clash tutorial.

christiaanb, 2017-03-13:
You might not be able to synthesize the functions in those libraries to VHDL though, because:

1. They might use I/O or other primitive types not supported by Clash
2. GHC didn't store the intermediate representation for some of the functions (it only goes this for function it thinks it might inline at some point in time).

So, yes, you can use those other libraries, but you might not be able to translate them to VHDL.