clash-lang/clash-compiler #194: Tutorial (FIR.hs) does not work

Aruee, 2017-01-12:
Hey,

I hope that this hasn't been reported before or is a known issue - I couldn't really find anything at first glance. I'm just trying to get started with clash, but even just going through the tutorial fails when running the FIR.hs example:

```
$ clash FIR.hs 
[1 of 1] Compiling FIR              ( FIR.hs, FIR.o )

FIR.hs:15:27:
    Could not deduce (KnownNat (n + 1)) arising from a use of ‘bundle’
    from the context (Default a, KnownNat n, SaturatingNum a)
      bound by the type signature for
                 fir :: (Default a, KnownNat n, SaturatingNum a) =>
                        Vec (n + 1) a -> Signal a -> Signal a
      at FIR.hs:(11,8)-(12,44)
    In the second argument of ‘(<$>)’, namely ‘bundle xs’
    In the expression: dotp coeffs <$> bundle xs
    In an equation for ‘y_t’: y_t = dotp coeffs <$> bundle xs
$ clash --version
CAES Language for Synchronous Hardware, version 0.6.24 (using clash-lib, version: 0.6.21)
$ cabal --version
cabal-install version 1.24.0.2
compiled using version 1.24.2.0 of the Cabal library 
$ ghc --version
The Glorious Glasgow Haskell Compilation System, version 7.10.3
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial
```

While I assume that the error is somewhere on my side and will keep trying, I though I'd let you know that the tutorial seems to be broken at least for some systems.

christiaanb, 2017-01-12:
@Aruee Thanks for the report. It seems I updated the link on the http://clash-lang.org front page, but never update the one in the tutorial. The file you should be using is: https://raw.githubusercontent.com/clash-lang/clash-compiler/0.6/examples/FIR.hs

I'll try to fix the tutorial soon.

Aruee, 2017-01-12:
Works perfectly, thank you! I'll leave this open as a reminder for you to fix the tutorial though ;)

christiaanb, 2017-01-17:
This has now been fixed with the release of CLaSH 0.7

closed 2017-01-17 by christiaanb