clash-lang/clash-compiler #195: Build failure with GHC 8.0.1 for Stackage Nightly

snoyberg, 2017-01-17:
```
> /tmp/stackage-build12$ stack unpack clash-ghc-0.7
Unpacked clash-ghc-0.7 to /tmp/stackage-build12/clash-ghc-0.7/
> /tmp/stackage-build12/clash-ghc-0.7$ runghc -clear-package-db -global-package-db -package-db=/var/stackage/work/builds/nightly/pkgdb Setup configure --package-db=clear --package-db=global --package-db=/var/stackage/work/builds/nightly/pkgdb --libdir=/var/stackage/work/builds/nightly/lib --bindir=/var/stackage/work/builds/nightly/bin --datadir=/var/stackage/work/builds/nightly/share --libexecdir=/var/stackage/work/builds/nightly/libexec --sysconfdir=/var/stackage/work/builds/nightly/etc --docdir=/var/stackage/work/builds/nightly/doc/clash-ghc-0.7 --htmldir=/var/stackage/work/builds/nightly/doc/clash-ghc-0.7 --haddockdir=/var/stackage/work/builds/nightly/doc/clash-ghc-0.7 --flags=
Configuring clash-ghc-0.7...
> /tmp/stackage-build12/clash-ghc-0.7$ runghc -clear-package-db -global-package-db -package-db=/var/stackage/work/builds/nightly/pkgdb Setup build
Building clash-ghc-0.7...
Preprocessing executable 'clash' for clash-ghc-0.7...

In file included from src-bin/GHCi/UI.hs:33:0: error: 

/opt/ghc/8.0.1/lib/ghc-8.0.1/ghc-8.0.1/include/HsVersions.h:23:0: error:
     fatal error: ../includes/ghcautoconf.h: No such file or directory
     #include "../includes/ghcautoconf.h"
     ^
compilation terminated.
`gcc' failed in phase `C pre-processor'. (Exit code: 1)
```

christiaanb, 2017-01-17:
Thank you very much for the notification @snoyberg . This has now been fixed by 1c25031672b8fdadbbbcc08759e3fd3864bb5108, which is included in the 0.7.0.1 release on hackage: http://hackage.haskell.org/package/clash-ghc-0.7.0.1

closed 2017-01-17 by christiaanb

snoyberg, 2017-01-17:
Awesome, thanks for the quick fix.