clash-lang/clash-compiler #201: CλaSH needlessly copies hardware

strake, 2017-01-29:
Consider a part of a very simple processor core:

    {-# LANGUAGE PatternSynonyms #-}

    module Test where

    import CLaSH.Prelude.Safe hiding (Word)

    type Word = BitVector 32

    instruct :: BitVector 16 -> Maybe (Word -> Word -> Word)
    instruct i = case slice d1 d0 i of
        RegReg -> aluMay
        RegImm -> (\ alu -> pure . flip alu imm) <$> aluMay
        _ -> Nothing
      where imm :: Word
            imm = resize (slice d15 d9 i)

            aluMay :: Maybe (Word -> Word -> Word)
            aluMay = case (i ! 2, i ! 0 .&. i ! 12) of
                (Add, 0) -> Just (+)
                (Add, 1) -> Just (-)
                (Xor, 0) -> Just xor
                _ -> Nothing

    pattern RegReg = 0
    pattern RegImm = 1

    pattern Add = 0
    pattern Xor = 1

    topEntity :: BitVector 16 -> Word -> Word -> Maybe Word
    topEntity = (sequenceA .) . sequenceA . instruct

I want this to emit 1 ALU and a mux on one operand input between the given operand and the immediate value `imm`, thus (pseudocode):

    answer = case (i[2], i[0] & i[12]) of
        (0, 0) -> u + v
        (0, 1) -> u - v
        (1, 0) -> u ^ v
        _ -> _

    u = x
    v = case slice d1 d0 i of
        0 -> y
        1 -> imm

    imm = resize (slice d15 d9 i)

But rather it emits a mux and 2 ALUs, thus:

    answer0 = case (i[2], i[0] & i[12]) of
        (0, 0) -> x + y
        (0, 1) -> x - y
        (1, 0) -> x ^ y
        _ -> _

    answer1 = case (i[2], i[0] & i[12]) of
        (0, 0) -> x + imm
        (0, 1) -> x - imm
        (1, 0) -> x ^ imm
        _ -> _

    answer = case slice d1 d0 i of
        0 -> answer0
        1 -> answer1

    imm = resize (slice d15 d9 i)

which is wasteful
[Test.tar.txt](https://github.com/clash-lang/clash-compiler/files/737771/Test.tar.txt)

.

Ericson2314, 2017-01-29:
One of the things I like about CLaSH is that its low-level enough to intuit cost pretty well. I'd *generally* expect a function to instantiated once per application (per callee's application, recursively up to `topLevel`)---there are critical path length vs. area trade-offs here so the optimization isn't so clear-cut.

Granted, types like  `Maybe (Word -> Word -> Word)` (to me at least) have a less obvious a cost model anyways so my statement may not so cleanly apply.

In your specific case I don't think it would be too hard to restructure your code to achieve the desired result explicitly? Something like the following seems clear to me:
```haskell

instruct :: BitVector 16 -> Maybe (Word -> Word -> Word)
instruct i = f <$> aluMay
 where 
   f = case slice d1 d0 i of
    RegReg -> id
    RegImm -> (\ alu -> \ x y -> alu x imm)
```

comonoidial, 2017-01-29:
I have not looked at the exact transformations CLaSH performs on this example, but I think the problem is caused by wrapping functions in the Maybe datatype.
Higher order or wrapped functions are not representable in hardware, so CLaSH tries aggressively to remove them all. This usually involves specialisation of functions which can duplicate code.

I think CLaSH could be made more careful with duplication, but improving the transformation process is tricky (without making it slow or reject other code).

Changing things of type: `Maybe (Word -> Word -> Word)`  into  `Word -> Word -> Maybe Word` may be a workaround for this duplication.

christiaanb, 2017-01-29:
As @Ericson2314 and @comonoidial indicate, normally, functions are instantiated every application. With this high-order value, several things are going on though. First of all, let/where-bound functions are usually inlined, instead of being lifted to a global function. https://github.com/clash-lang/clash-compiler/blob/8821c79d231cbdd2adcecbaf79667902cdd279cb/clash-lib/src/CLaSH/Normalize/Strategy.hs#L145 explains why.

Now, for functions occuring the alternatives for case statements we do have the DEC transformation, which is described here: https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/CLaSH/Normalize/DEC.hs#L11. And here, https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/CLaSH/Normalize/DEC.hs#L379, is the function that determines whether it thinks a functions is interesting to lift out of a case-alternative.

This selection function is currently rather simplistic, and only checks based on the function itself, and not so much its context. Now, as @Ericson2314 indicates, there is a trade-off to be made as to whether to lift a function out of a case-alternative or not, because it introduces a multiplexer on the input, which increases the propagation delay. Also, and this holds especially for FPGAs, this multiplexer might actually be more expensive in terms of area than the function that is being lifted out, so it makes more sense to leave it duplicated. Now, in your example code, this input selection multiplexer can be shared amongst several lifted primitives, and so the heuristic as to whether to lift the primitive our of the case-alternative is failing. So yeah, there is certainly room to improve the heuristic, so I'll definitely leave this issue open.

strake, 2017-01-30:
A-ha, yes, when i add `(+#)`, `(-#)`, `xor#` to the list of "interesting" primitives, it indeed multiplexes the input.

A heuristic which would work in this situation, and i believe not make the code worse in any situation, is to lift if only one argument of the function varies across cases, and that argument is no wider than the case expression itself.

strake, 2017-01-30:
Could we somehow let the user mark a term as "interesting"?

strake, 2017-01-30:
Ericson2314: This has nonsame semantics; it would rather need to be `aluMay >>= f` for some `f`, which i tried, and which also copies the ALU.

comonoidial: I tried this; it emitted the ALU as its own module, but duplicated that module!

Ericson2314, 2017-01-30:
Strake ah right I removed a Nothing case. Yeah I agree with the idea of explicit interesting—while optimizations are nice, I want everything to be (at least unsafely) expressible without them.

christiaanb, 2017-01-30:
@strake Functions's could be marked using `{-# ANN ... #-}` pragmas. Something that could be implemented: 

1. Make all instances of `CLaSH.Sized.Internal.Unsigned.+#` within one module interesting by having the following annotation:

```haskell
{-# ANN CLaSH.Sized.Internal.Unsigned.+# IsInterestingToLift #-}
```

present in that module.

2. And this would be the smallest granularity possible, annotate a function `f` with:

```haskell
{-# ANN f  (IsInterestingToLiftInside [''CLaSH.Sized.Internal.Unsigned.+#] #-}
```

Where `'CLaSH.Sized.Internal.Unsigned.+#` would only be considered interesting when used inside `f`.

comonoidial, 2017-01-30:
I think the logic in `interestingToLift` is the wrong way around, it is now trying to enumerate things more expensive than input multiplexers. This leads a never ending job of annotating new primitives.

What about lifting everything by default, unless it is know to be more trivial than a multiplexer?

In my opinion making the tradeoff between duplication and multiplexer is the job of the synthesis tool. And I think the synthesis optimisations having a better chance at introducing a duplication that reduces a critical path, than finding complex duplicated things that can better be unified.

And from a user perspective duplication is a surprising thing, that is hard to predict and not easy to get control over if you want to. On the other hand when more duplication is needed for a faster circuit, it is just a matter of writing code in lower level more explicit style.

Ericson2314, 2017-01-30:
@comonoidial that's interesting. It sure would be nice if GHC supported `{-# INLINE #-}` pragmas on function applications, not just function definitions

I've been trying to think of a hardware model for things like `Maybe (a -> b)` / function fields in an ATD. It's *possible* that they can be thought of as circuits (the functions) whose use is guarded by the discriminant, i.e. one must mux to use the output.

[In fact, this made me fantasize of a `PowerGate#` isomorphic to `Maybe` causing the circuit to be powered off during the `Nothing` case. I don't really know much about power gating, and I suspect Verilog/VHDL doesn't handle it outside of some horrible vendor-specific hack.]

strake, 2017-01-31:
> It's possible that they can be thought of as circuits (the functions) whose use is guarded by the discriminant, i.e. one must mux to use the output.

Indeed, this is my model. It corresponds to `sequenceA` effectively.

PR #1014 (2020-01-16): `Clash.Magic.deDup`: Force deduplication
Fixes #201
commit 1fa2722a (2020-01-16): `Clash.Magic.deDup`: Force deduplication

fixes #201
closed 2020-01-16 by christiaanb