clash-lang/clash-compiler #202: fold tree error in VHDL

DigitalBrains1, 2017-02-09:
The following code generates problematic VHDL:
```haskell
import CLaSH.Prelude

topEntity :: Signal(Vec 10 Bit) -> Signal Bool

topEntity = mealy f ()
    where
        f () i = ((), i == repeat 0)
-- This works:
--        f () i = ((), bitCoerce i == (0 :: Unsigned 10))
```

Quartus (11.1) errors out with:
```
Error (10384): VHDL assignment error at main_topentity1.vhdl(58): index 19 is outside the range (0 to 18) of object "intermediate"
```

Looking at the [generated VHDL](https://gist.github.com/DigitalBrains1/2ecb7fc2bdcdfe56c16e060e14314680) for the binary tree construction, I notice that `levels = 4`, and `d_0` ranges from 0 to 3. This means that `depth2index` returns values up to 30 for `depth2index(5, 1)`, yet `intermediate` has only 19 elements.

christiaanb, 2017-02-09:
That's odd, it seems that:
https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/CLaSH/Normalize/Transformations.hs#L1020

isn't triggered. Because it should only generate the VHDL you provided if the `Vec` has a power of two length.

DigitalBrains1, 2017-02-09:
Does it do the right thing for you?

I encounter this with CλaSH installed per the instructions on [clash-lang.org](http://clash-lang.org/) on the noc1 machine of CAES. It's a user install of GHC 8.0.2 from `ghc-8.0.2-x86_64-centos67-linux.tar.xz`, and a compile from source for cabal-install from `cabal-install-1.24.0.2.tar.gz` since the libc version didn't match on the binary download.

christiaanb, 2017-02-09:
It seems that

```
topEntity :: Vec 10 Bit -> Bool
topEntity = (== repeat 0)
```

is enough to trigger it.

The problem is that (`==`) for vector is desugered, by GHC, to:

```
\a b -> fold @Bool @(Any Nat) (zipWith @Bit @Bit @10 a b)
```

where the problem is the `@(Any Nat)`, instead of `@10`, meaning Clash cannot figure out the size of the fold at the normalisation stage.

christiaanb, 2017-02-09:
Fixed by 860cfdabfd99b998913eea44c17e8f1853469f8b, thanks for the report.

closed 2017-02-09 by christiaanb