clash-lang/clash-compiler #215: clash-ghs-0.7.1 won't install over  clash-prelude-0.11

ggreif, 2017-04-12:
I did a `cabal update` and `cabal install clash-ghc` over an existing v0.7/v0.11 installation.

Resulted in this:
```
[ 6 of 13] Compiling CLaSH.GHC.LoadInterfaceFiles ( src-ghc\CLaSH\GHC\LoadInterf
aceFiles.hs, dist\build\CLaSH\GHC\LoadInterfaceFiles.o )

src-ghc\CLaSH\GHC\LoadInterfaceFiles.hs:23:1: error:
    Failed to load interface for `CLaSH.Annotations.Primitive'
    Perhaps you meant
      CLaSH.Annotations.TopEntity (from clash-prelude-0.11)
    Use -v to see a list of the files searched for.
cabal: Leaving directory 'C:\Users\ggreif\AppData\Local\Temp\cabal-tmp-28892\cla
sh-ghc-0.7.1'
cabal: Error: some packages failed to install:
clash-ghc-0.7.1 failed during the building phase. The exception was:
ExitFailure 1
```

Probably [this line](https://github.com/clash-lang/clash-compiler/blob/0.7/clash-ghc/clash-ghc.cabal#L110) needs to be adapted to require `clash-prelude-0.11.1`.

ggreif, 2017-04-12:
Also that error shouldn't have happened with a [PVP abiding](https://pvp.haskell.org/faq/) version number. I guess such a breaking change would necessitate `clash-ghc-0.8`, but I digress...

christiaanb, 2017-04-12:
Thanks for the report. I've update the `.cabal` file on hackage: http://hackage.haskell.org/package/clash-ghc-0.7.1/clash-ghc.cabal

christiaanb, 2017-04-14:
Also fixed the `.cabal` files in the source repo.

closed 2017-04-14 by christiaanb