clash-lang/clash-compiler #216: Couldn't match type ‘'False’ with ‘'True’

riaqn, 2017-04-14:
I guess this is mostly some dependent types issue, but I really don't know where goes wrong. Here is the source code:
```
module TopEntity where

import CLaSH.Prelude

import Data.Bool

import qualified Data.List as L

data Command = Destruct Object
             | Construct Object Alt

data Object = Term
            | Modifiers
            | Args
            | Backtrace

type Addr = BitVector 25
type Ptr = BitVector 32
type Alt = BitVector 3
type Reg = Index 8

-- 32MB of memory
memDepth = d25
-- every word(Ptr) is 4 bytes
wordDepth = d2

data Stage = GC
           | Idle
           | Cons Object Alt (Index 4)
           | Des Object Bool (Index 4) (Index 4)

data State = State { stage :: Stage
                   , regs :: Vec 8 Ptr
                   , heap :: Addr
                   }

mapping :: Object -> (Reg, Alt -> (Index 4, Vec 4 Reg))
mapping obj = case obj of
  Term -> (0, \alt -> case alt of
              0 -> (1, $(listToVecTH [6, 0, 0, 0 :: Index 8]))
              1 -> (2, $(listToVecTH [0, 4, 0, 0 :: Index 8]))
              2 -> (1, $(listToVecTH [0, 0, 0, 0 :: Index 8]))
              3 -> (3, $(listToVecTH [6, 0, 4, 0 :: Index 8]))
              4 -> (3, $(listToVecTH [6, 7, 0, 0 :: Index 8]))
          )
  Modifiers -> (1, \alt -> case alt of
                   0 -> (0, $(listToVecTH [0, 0, 0, 0 :: Index 8]))
                   1 -> (3, $(listToVecTH [6, 0, 1, 0 :: Index 8]))
                   2 -> (3, $(listToVecTH [6, 7, 1, 0 :: Index 8]))
               )
  Args -> (2, \alt -> case alt of
              0 -> (0, $(listToVecTH [0, 0, 0, 0 :: Index 8]))
              1 -> (2, $(listToVecTH [4, 2, 0, 0 :: Index 8]))
          )
  Backtrace -> (3, \alt -> case alt of
                   0 -> (0, $(listToVecTH [0, 0, 0, 0 :: Index 8]))
                   1 -> (4, $(listToVecTH [6, 1, 4, 3 :: Index 8]))
               )

-- Term [Modifier] [Term] [BT]
-- Term ?? Index Index
topEntity :: Signal (Maybe Command, Bool, Maybe Ptr, Bool) -> Signal (Bool, Maybe Ptr, Bool, Maybe (Either (Addr) (Addr, Ptr)))
topEntity = mealy (\s (cmdI, cmdOfull, memI, memOfull) -> case (stage s) of -- out: (cmdIreq, cmdO, memIreq, memO)
                 GC -> ( s
                       , (False, Nothing, False, Nothing)
                       )
                 Idle -> let (cmdIreq, s') = case cmdI of
                               Nothing -> (False, s)
                               Just (Construct obj alt) -> ( True
                                                           , s { stage = Cons obj alt 0
                                                               }
                                                           )
                               Just ( Destruct obj) -> ( True
                                                       , s { stage = Des obj False 0 0}
                                                       )
                         in
                           ( s'
                           , (cmdIreq, Nothing, False, Nothing)
                           )
                 Cons obj alt memOn -> let (heap', regs', stage', memO) =
                                             if memOn == len then (heap s + (fromInteger $ toInteger $ shiftL len 2), replace x (fromInteger $ toInteger $ heap s) (regs s), Idle, Nothing)
                                             else if memOfull then (heap s, regs s, stage s, Nothing)
                                             else (heap s, regs s, Cons obj alt memOn_, Just $ Right (heap s + (fromInteger $ toInteger $ shiftL memOn 2),  (regs s) !! (args !! memOn)))
                                      in
                                        ( s { stage = stage'
                                            , heap = heap'
                                            , regs = regs'
                                            }
                                        , (False, Nothing, False, memO)
                                        )
                   where (x, args_) = mapping obj
                         (len, args) = args_ alt
                         memOn_ = memOn + 1
                 Des obj ret memOn memIn -> let (memO, memOn') =
                                                    if memOfull || memOn == len then (Nothing, memOn)
                                                    else (Just $ Left $ shiftL (addr + (fromInteger $ toInteger memOn)) 2, memOn_)
                                                (memIreq, memIn', regs') =
                                                  case memI of
                                                    Nothing -> old
                                                    Just memIq -> if memIn == len then old else
                                                      (True, memIn_, replace x memIq (regs s))
                                                  where old = (False, memIn, regs s)
                                                (cmdO, ret') =
                                                    if cmdOfull || ret then (Nothing, ret)
                                                    else (Just $ fromInteger $ toInteger $ alt, True)
                                            in
                                                ( s { stage = if memIn == len && ret then Idle else Des obj ret' memOn' memIn'
                                                    , regs = regs'
                                                    }
                                                , (False, cmdO, memIreq, memO)
                                                )
                   where (x, args_) = mapping obj
                         reg = (regs s) !! x
                         alt = (slice d31 d29 reg)
                         addr = slice (memDepth `subSNat` wordDepth `subSNat` d1) 0 reg
                         (len, args) = args_ alt
                         memOn_ = memOn + 1
                         memIn_ = memIn + 1
             )
            (State { heap = 0
                   , regs = repeat 0
                   , stage = Idle
                   })
```

The error message:
```
TopEntity.hs:110:74: error:
Couldn't match type ‘'False’ with ‘'True’
arising from a use of ‘memO’
Failed, modules loaded: none.
```

christiaanb, 2017-04-14:
Yeah... the error message is quite cryptic.
The problem is:

```
addr = slice (memDepth `subSNat` wordDepth `subSNat` d1) 0 reg
```

if you change that do:

```
addr = slice (memDepth `subSNat` wordDepth `subSNat` d1) d0 reg
```

You will get a different error though, which is less cryptic:

```
examples/TopEntity.hs:65:13: error:
    • Couldn't match type ‘23’ with ‘25’
      Expected type: Signal (Maybe Command, Bool, Maybe Ptr, Bool)
                     -> Signal (Bool, Maybe Ptr, Bool, Maybe (Either Addr (Addr, Ptr)))
        Actual type: Signal (Maybe Command, Bool, Maybe Ptr, Bool)
                     -> Signal
                          (Bool, Maybe Ptr, Bool, Maybe (Either (BitVector 23) (Addr, Ptr)))
...
```

riaqn, 2017-04-14:
@christiaanb wow! Thanks for the lightning reply! I will try that.

Now that we've come to it, have the developers considered moving to languages with native dependent types(like idris)? At least the error messages would be more friendly.

christiaanb, 2017-04-14:
Normally the error messages shouldn't be this bad... this just seems to be an unlucky interaction with one of the type-checker plugins and a too strict evaluation of a type-function. (An error like `couldn't mactch True with (25 <= 23)` would have been nices) 

We have considered a using a dependently-typed language as an input language, but have not explored it further because:
* The current translation theory is heavily based on System F, so moving towards a dependently-typed language would mean new research, and a complete reimplementation of the compiler.
* Describing feedback loops in clash critically depends on laziness, we don't know if Idris' laziness annotations would work well for us
* It is really nice to have the entire Haskell library ecosystem at your disposal during simulation
* We haven't found compelling use-cases for using dependent types in everyday industrial practice beyond what is currently already available in Haskell
* etc.

closed 2021-06-02 by alex-mckenna