clash-lang/clash-compiler #219: Verilog for concatenations involving sum types is sometimes wildly wrong

cbiffle, 2017-05-03:
On CLaSH 0.7.2, Verilog output (and probably SystemVerilog output) for certain constructs involving a sum type `t` can accidentally concatenate a platform-Int-sized bitstring in place of a `BitSize t`-size bitstring.

I haven't found the precise cause and I don't have a fix yet.  I don't actually even have a reliable reduction case (see below).

Here is an example of the offending output:

```verilog
  output [31:0] result;
  wire signed [63:0] result_app_arg;
  assign result_app_arg = $signed(64'sd8);
  
  assign result = {({({({({({7'b0000000,(5'd1)}),(5'd0)}),3'b010}),(5'd0)}),($unsigned(result_app_arg))}),2'b11};
```

`result` is a 32-bit value, and in Haskell it receives a `BitVector 32` result correctly.  In Verilog, however, it's being assigned from a **91-bit value**.  This is legal in Verilog, because Verilog hates you, but that's beside the point.  We wind up taking the bottom 32 bits of a 91 bit value, which are wrong.

The problem is `result_app_arg`.  It is the result of the following structure:

```haskell
data Opcode = .... | OpStore | ...
instance BitPack Opcode where
  type BitSize Opcode = 5
  pack = fromIntegral . fromEnum

something = pack ( ... OpStore ... )
```

`OpStore`'s enum value is 8, so `fromEnum OpStore` is giving the `64'sd8` value you see above... and then the 64-bit Int version is concatenated instead of the 5-bit version.

I believe the problem is [here in the backend](https://github.com/clash-lang/clash-compiler/blob/master/clash-verilog/src/CLaSH/Backend/Verilog.hs#L434-L435).  If you compare to [the equivalent VHDL backend code](https://github.com/clash-lang/clash-compiler/blob/master/clash-vhdl/src/CLaSH/Backend/VHDL.hs#L812-L816), you'll note that the VHDL version has a `resize` operation to explicitly truncate the value.  The Verilog is missing this.

Is the fix a simple matter of appending an index selector to the Verilog expression (the closest thing we've got to VHDL resize)?  Maybe.  I'll try in a bit.

Here's the really infuriating part though: it doesn't reproduce reliably.  Reproducing this depends on *whether all the code is in one Haskell module or not.*  When I move everything into one module to post a nice clean reduction for the issue report... the constants get magically inlined and the output stops breaking.

Here's a two-module reduction.

Main.hs:

```haskell
import CLaSH.Prelude hiding (v)
import qualified Bork2 as A

lsbs :: BitVector 2
lsbs = 0b11

-- This function is identical to the one over in Bork2, but if you use this
-- one instead of A.rtype, everything starts working.
rtype :: A.Opcode -> BitVector 3 -> BitVector 7 -> Unsigned 5 -> Unsigned 5 -> Unsigned 5 -> BitVector 32
rtype opc f3 f7 rd rs1 rs2 = pack (f7, rs2, rs1, f3, rd, opc, lsbs)

topEntity :: Signal (BitVector 32)
topEntity = pure $ A.rtype A.OpStore 2 0 0 0 1
--topEntity = pure $ rtype A.OpStore 2 0 0 0 1  -- works
```

Bork2.hs:

```haskell
module Bork2 where

import CLaSH.Prelude hiding (v)

data Opcode = OpLoad   | OpLoadFp  | OpCust0 | OpMiscMem
            | OpImm    | OpAuiPc   | OpImm32 | Op48Bit
            | OpStore  | OpStoreFp | OpCust1 | OpAmo
            | OpOp     | OpLui     | OpOp32  | Op64Bit
            | OpMadd   | OpMSub    | OpNMSub | OpNMAdd
            | OpOpFp   | OpRes1    | OpCust2 | Op48Bit2
            | OpBranch | OpJalr    | OpRes2  | OpJal
            | OpSys    | OpRes3    | OpCust3 | Op80Bit
            deriving (Enum)

instance BitPack Opcode where
  type BitSize Opcode = 5
  pack = fromIntegral . fromEnum
  unpack = toEnum . fromIntegral

lsbs :: BitVector 2
lsbs = 0b11

rtype :: Opcode -> BitVector 3 -> BitVector 7 -> Unsigned 5 -> Unsigned 5 -> Unsigned 5 -> BitVector 32
rtype opc f3 f7 rd rs1 rs2 = pack (f7, rs2, rs1, f3, rd, opc, lsbs)
```

cbiffle, 2017-05-03:
My initial assumption about the location of the issue was wrong.

It seems that the faulty code is always coming from the `CLaSH.Sized.Internal.BitVector.fromInteger#` primitive, specifically [here](https://github.com/clash-lang/clash-compiler/blob/master/clash-verilog/primitives/CLaSH.Sized.Internal.BitVector.json#L241).  The rough shape of the problem is the same: [the VHDL equivalent](https://github.com/clash-lang/clash-compiler/blob/master/clash-vhdl/primitives/CLaSH.Sized.Internal.BitVector.json#L402) has an explicit resize, and the Verilog does not.  Adding an index range select, either inside or outside the `$unsigned`, fixes this particular manifestation.  The counterpart `toInteger#` is not affected because it's a declaration template assigning to an explicitly sized wire result, so Verilog happily (and mindlessly) strips off extra bits.

But this implies that other expression templates could be similarly affected in Verilog.

I'm going to (1) read over the other expression primitives to look for similar discrepancies between Verilog and VHDL, (2) ensure that my Verilog and Haskell test benches are matching each other again, and (3) send a pull request.

christiaanb, 2017-05-03:
Thanks for the thorough analysis, and the fix.

There are ideas/plans to mitigate issues like this in the future:
- Adding compile-time evaluation for all clash primitives would ensure that `((fromInteger# . toInteger#) 8)` would be fully evaluated to the correct literal. (Currently the evaluator is incomplete)
- I plan to add one day add bit-width inference for  `Integer`, so that clash stops encoding `Integer` to 64-bit signed integers, and will use a proper bit-width. At the end of http://hackage.haskell.org/package/clash-prelude-0.11.2/docs/CLaSH-Tutorial.html#g:19 it therefore says that you should currently use `Integer` with due diligence.

cbiffle, 2017-05-03:
*I've changed the title to (1) make this slightly more general and (2) point out that the problem here is really with Verilog, not CLaSH. ;-)*

I've completed a more detailed review of the primitives, and I don't see any clear remaining errors in this area.  (Also my tests are passing.)

There's a subtlety in how some of the primitives are defined, and I think it's worth mentioning here (as it is not documented elsewhere).

Verilog-2001 will select the bit width of an expression based not only on its operands, but on its context.  In particular, when an expression appears in an assignment, the bit width of not only the outermost expression value, *but all its internal intermediate values*, can get extended to match the size of the LHS.  This occurs in every expression position where the size is not *self-determined.*

(I managed to track down a copy of the '01 spec [here](https://inst.eecs.berkeley.edu/~cs150/fa06/Labs/verilog-ieee.pdf).  The rules in question are in section 4.4.1 on pages 59/60.)

An important self-determined position that is *not* listed explicitly there is function/task argument.  That is, the result of a function may get extended to the width of the LHS when used in an assignment, but its arguments are unaffected (transitively).

In particular, the functions `$signed` and `$unsigned` are explicitly defined (section 4.5) as producing a value the same width as their argument.  This means that `$unsigned(x)` can be used as a general pattern to stop bit-width expansion from creeping into the expression `x`.

And CLaSH is doing exactly this.  In particular, functions that explicitly extend or truncate values, such as `resize#` for `BitVector`, use `$unsigned` or `$signed` as appropriate, and all the uses look correct.

Without this we could easily get incorrect results from any expression whose operands are not completely "self-determined" per the spec.  Like, say, shifts.  This naive translation:

```haskell
x = 2 :: BitVector 3
y = (zeroExtend $ x `shiftL` 2) :: BitVector 7
```

```verilog
wire [2:0] x = 3'b2;
wire [6:0] y = x << 2;  // Handle zeroExtend implicitly
```

...is wrong in Verilog-2001, because it implicitly extends `x` to 7 bits *before* the shift.  CLaSH uses this translation (simplified here):

```verilog
wire [2:0] x = 3'b2;
wire [6:0] y = $unsigned(x << 2);
```

...which determines the bit-width of `x` without consideration of the assignment context and produces the correct result.

So at this point I'm confident in the state of things.  We definitely could use some regression tests on this though...

closed 2017-05-03 by cbiffle