clash-lang/clash-compiler #229: Can't get the clash REPL installed

gameboo, 2017-05-22:
Following the install steps described at http://www.clash-lang.org/ or at http://hackage.haskell.org/package/clash-prelude-0.11.2/docs/CLaSH-Tutorial.html, in particular the `cabal install clash-ghc --enable-documentation --enable-executable-dynamic` step, I get an install of clash that allows me to load the MAC tutorial script in `ghci` an for it to behave as expected. However, it doesn't get me the `clash` REPL binary:
```
$ clash --interactive
No command 'clash' found, did you mean:
[...]
```

I followed the steps on a machine with a recent Ubuntu install:
```
$ uname -a
Linux gb-xeons 4.10.0-21-generic #23-Ubuntu SMP Fri Apr 28 16:14:22 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

The github issue tracker is the only place I could find to ask that question. If there is another more appropriate place, I am happy to go ask my question again over there.

Am I missing something obvious ?

christiaanb, 2017-05-23:
It's easy to miss in the installation instructions, but did you add `$HOME/.cabal/bin` to your `PATH`?

gameboo, 2017-05-23:
I did not !
I already had ghc 8.0.2 installed and naively though that this was required only for this step.
Do you believe it's worth repeating this instruction to the "install clash" step, or lifting it out of the install ghc step, for silly people like me ?

Thank you for your fast answer.

dented42, 2017-06-13:
An alternative (if you are using stack to manage your projects) is to let stack manage all the path stuff for you. If you do `stack install clash-ghc` then you can invoke clash with `stack exec -- clash --interactive`.

martijnbastiaan, 2019-12-20:
Closing this issue - we've added instructions to our website / github.

closed 2019-12-20 by martijnbastiaan