clash-lang/clash-compiler #231: `inout` signals

dented42, 2017-06-13:
It'd be nice if Clash could generate entities with `inout` signals as it reduces the amount of outside vhdl glue that is required. I can think of two potential ways of modelling them. The first seems a little aesthetically displeasing and I'm unsure if the second is sound. Perhaps one of them might lead somewhere worth going?

Despite the second approach feeling nicer to me, my gut instinct is that the first approach is more robust.

# `inout` as a signal type
The first way to model them would be to have a variant of `Signal` that has a symbol type parameter that is used to determine when a function has inputs and outputs that should map to the same `inout` signal. If `Signal` and `Signal'` are to be specialisations of this `InoutSignal` type then using the empty string for the symbol parameter should probably be interpreted as that signal being unique; otherwise devising a new symbol for each signal will be annoying.

 As an example this would mean that the type of a generic tristate I/O buffer (where high impedance maps to `undefined`) would be something like `Signal Bool -> Signal a -> InoutSignal "io" a -> (InoutSignal "io" a, Signal Bool)` or maybe `KnownSymbol s => Signal Bool -> Signal a -> InoutSignal s a -> (InoutSignal s a, Signal Bool)`.

This would probably work but having to muck about with symbols seems kinda gross to me.

# `inout` as a function
 Another way might be to model them as functions. An `inout` signal in VHDL of type `a` would correspond to the Clash type `Signal a -> Signal a`.

This would make the type of a top entity with a single `inout` bit signal `Signal Bit -> Signal Bit`, which conflicts with the current interpretation as an entity with one `in` and one `out` signal. For this to work there should probably be a data type that the compiler recognises that gets translated to an `inout` signal in VHDL? Presumably something along the lines of:

```
newtype Inout' clk a = Inout {
  inout :: Signal' clk a -> Signal' clk a
}

type Inout a = Inout' SystemClock a
```

Unfortunately I'm not sure if the semantics of this interpretation can be consistent. If a function/entity has some input and output signals then should the `Inout` function be an argument or result? Should the previous tristate I/O buffer example have the type `Signal Bool -> Signal a -> Inout a -> Signal Bool` or the type `Signal Bool -> Signal a -> (Inout a, Signal Bool)`? Neither seems terribly satisfactory and I suspect the former gives GHC the opportunity to optimise away the `inout` function's application is certain cases.

expipiplus1, 2017-06-14:
This problem becomes much simpler if you allow `inout` signals to be opaque values which can only be produced at the top level and only consumed by primitives.

dented42, 2017-06-15:
True... But wouldn't that make them impossible to test inside haskell/clash? And as opaque values it seems like it would still suffer from the argument vs result dilemma. Either the `inout` value is an input or output to a function, I don't think it can be both without some type level way of matching inputs and outputs so that clash knows to compile them into the same signal.

christiaanb, 2017-06-15:
I think I could change the TopEntity annotations to support tristate drivers for inout ports. Honest-to-goodness inout like you have them in VHDL is going to be really difficult. In Vhdl we have resolution of signals, meaning the simulator looks for all the drivers of a signal, resolves it to a value (0,1,Z,X), and then a delta delay later, will resolve other signals dependent on the value. We cannot have this behavior in clash, because we only have normal function semantics; lazy at that. Even using I/O refs, some sort of outside scheduler, I really don't know how to get the dual driver resolution working like you have it in Vhdl.

dented42, 2017-06-15:
I'm unsure what form such an annotation would take, but it would definitely solve my particular use case.

martijnbastiaan, 2017-10-11:
For now we have settled on the simplest solution where we can mark a function input or output as Inout:

```haskell
topEntity :: Clock System Source
          -> Reset System Asynchronous
          -> Inout MemoryDom (BitVector 16) -- DRAM_DQ
          ...
```

It needs a review, but if all is well this will (probably) end up in Clash 0.99. It greatly reduces glue VHDL, but does suffer form an inability to be simulated and, as @dented42 noted, does not map cleanly to functions as these ports are neither inputs nor outputs. The former can be overcome by simply not using `inout`s for your own designs, while writing fronts for components having the need for `inout`s. This implementation does not offer a solution for the latter though.

We should probably move to something else in the future. I propose using `Maybe` types in combination with an implicit annotation based on port names...

```haskell
{-# ANN g
  (defTop{
    ....
    t_inputs = [ PortName "MY_INOUT_PORT" ],
    t_output = PortField "Output" [ PortName "MY_INOUT_PORT" ]
  })
  #-}
g :: Signal System (Maybe a) 
  -> Signal System (Maybe a)
g = ....
```

..or more explicit versions hereof. For simulation purposes we could make the user use a function checking if these to signals are not use independently, i.e.:

```haskell
g input = inout input output
    where
        output = .....
```

where

```haskell
inout (Just _) (Just _) = error "......."
inout _        o        = o
```

The upsides are that this is easy to simulate, easy to interpret (even for people new to Clash), and more closely represents how its implemented on an FPGA (AFAIK). The downsides are that it might rely on implicitness too much.

junjihashimoto, 2017-11-30:
👍

martijnbastiaan, 2018-08-03:
An implementation of inout ports, including an example, can now be found as a pull request: https://github.com/clash-lang/clash-compiler/pull/337. This implementation can be simulated and synthesized at the same time. This will hopefully land in the next release of Clash.

martijnbastiaan, 2018-08-08:
#337 has been merged. Closing this issue.

closed 2018-08-08 by martijnbastiaan