clash-lang/clash-compiler #235: SIMD targets for Clash?

chadbrewbaker, 2017-07-08:
Listening to Ed's ZuriHac talk, how hard would it be to goose Clash so that it emits AVX512 assembler?

Something as simple as adding two vectors would be extremely useful:

```c
void addVector (int a[], int b[], int c[], int size){
  int i;
  for(i=0;i<size;i++)
     c[i] = a[i]+b[i];
}
```

expipiplus1, 2017-07-08:
Link: https://www.youtube.com/watch?v=KzqNQMpRbac

christiaanb, 2017-07-09:
@chadbrewbaker I would rather emit something with more abstraction than AVX512 assembly; just like Clash emits VHDL/Verilog instead of an EDIF netlist (or even GDSII), so that we can get optimisations from compilers that are used to getting things down to the bare metal.

So clash could emit the C-code that you posted, along with all kinds of pragma's or whatnot so that the C-compiler will generate the AVX512 instructions.

In principle, this could be as simple as adding a C code-generator I guess... but it would have to rearrange the current IR, such that data-dependencies are executed in a sequential order; and probably a whole lot of other things.

chadbrewbaker, 2017-07-09:
The best target currently would be ISPC, http://ispc.github.io

It's C with a few constructs to denote parallelism.

Another target would be OpenCL C, which also compiles to FPGA, but is dodgy at getting use of SIMD units on CPUs.

thoughtpolice, 2017-07-12:
As a side note, I would find it thoroughly hilarious if Clash supported compiling to something like this. It's a very appropriate inverse to something like Vivado HLS which turns C into RTL, especially considering Haskell is nicer than C for hardware :)