clash-lang/clash-compiler #237: Not in normal form error when using complex type

josefschneider, 2017-07-19:
I have large complex type that I turn into a BitVector before assigning it to the output of my topEntity. Even though this hasn't been a problem for other similarly complex types in the past CLaSH refuses to convert this particular type into VHDL:

https://gist.github.com/anonymous/277354cd279ff47322cdac023877f0d7

In the CLaSH CLI I can create a ProblemType and convert it to a BitVector without a problem.

Thanks,
Josef

christiaanb, 2017-07-19:
Ah, I figured out what's going on, I haven't implemented one of the primitives (`CLaSH.Sized.Internal.BitVector.slice#` to be exact) in the compile-time evaluator in clash. Implementing these is a somewhat tedious job and I used do it on a bug-by-bug basis... However! we are currently working on implementing on all of the primitives in the compile-time evaluator. So this will hopefully be fixed soon.

As a work around, you could implement the `BitPack` instances for `Type1`, `Type2`, and `Type3` like so:

```
instance BitPack Type3 where
  type BitSize Type3 = 2
  pack x     = toEnum (fromEnum x)
  unpack x = toEnum (fromEnum x)
```

The generated VHDL will default to picking the last contstructor, `T3Val2`, in case you do unpack of a BitVector whose value is too large. But I think that is just as arbitrary as picking `T3Val0`. Also, you'll now get a run-time exception during simulation if the `BitVector` that you are converting with `unpack` has a value that is too large.

josefschneider, 2017-07-19:
Ah yes, that's a better way of doing it. All working now, thanks!

christiaanb, 2019-06-03:
Seems fixed in `master`

closed 2019-06-03 by christiaanb