clash-lang/clash-compiler #238: Nullary types break inliner

wyager, 2017-07-31:
Consider:

```
module Test where

import CLaSH.Prelude

data Void

voids :: Signal (Either Void Void)
voids = register (Left undefined) $ register (Right undefined) voids

e2b :: Either a b -> Bit
e2b (Left _) = 0
e2b (Right _) = 1

topEntity :: Signal Bit
topEntity = fmap e2b voids
```

Result:

```
: clash --verilog test.hs

test.hs:10:1: warning: [-Wtabs]
    Tab character found here, and in three further locations.
    Please use spaces instead.
Loading dependencies took 4.166565s
CLaSH.Normalize.Transformations(193): InlineNonRep: Test.topEntity_s11 already inlined 20 times in:Test.topEntity8214565720323806312
Type of the subject is: CLaSH.Signal.Internal.Signal'
  (CLaSH.Signal.Internal.Clk system 1000)
  (Data.Either.Either Test.X Test.X)
Function Test.topEntity8214565720323806312 will not reach a normal form, and compilation might fail.
Run with '-clash-inline-limit=N' to increase the inlining limit to N.
CLaSH.Normalize.Transformations(193): InlineNonRep: Test.topEntity_s11 already inlined 20 times in:Test.topEntity8214565720323806312
Type of the subject is: CLaSH.Signal.Internal.Signal'
  (CLaSH.Signal.Internal.Clk system 1000)
  (Data.Either.Either Test.X Test.X)
Function Test.topEntity8214565720323806312 will not reach a normal form, and compilation might fail.
Run with '-clash-inline-limit=N' to increase the inlining limit to N.

<no location info>: error:
    CLaSH error call:
    CLaSH.Normalize(183): Callgraph after normalisation contains following recursive cycles: [[Test.topEntity_s11]]
: clash --verilog test.hs
Loading dependencies took 4.28683s

<no location info>: error:
    CLaSH error call:
    CLaSH.Normalize(183): Callgraph after normalisation contains following recursive cycles: [[Test.voids8214565720323806156]]
```

If we replace `Void` with `Int` (or any other non-nullary type):

```
: clash --verilog test.hs
Loading dependencies took 4.606716s
Applied 40 transformations
Normalisation took 0.386403s
Netlist generation took 0.005289s
Testbench generation took 0.001879s
Total compilation took 5.004105s
```

The use case here is that I'm trying to come up with a standard signature for blocks in a CPU. By having unused signals be of type `Signal Void`, I can enforce at the type level whether or not a block uses a particular signal.

christiaanb, 2017-07-31:
The problem is that downstream synthesis tools deal poorly with zero-width things; so Clash considers them non-synthesisable. You can circumvent this behavior by running clash with: -clash-allow-zero-width

wyager, 2017-07-31:
Got it. As a stopgap, is there something that could be done with single-width wires? Like maybe synthesize any nullary constructor to a 1-bit wire with a const value? Synthesis tools might even be smart enough to optimize away that sort of thing.

commit a7e6f682 (2018-06-26): Zero-constructor data type needs 0 bits

We already had all the mechanisms in place to filter out 0-bit
data types from the generated VHDL, but we still handled
zero-constructor data types as untranslatable instead of just
translating them to 0-bit types.

Fixes #238
closed 2018-06-26 by christiaanb