clash-lang/clash-compiler #239: Declarative primitive with multiple outputs synchronized to different clocks

emakryo, 2017-08-02:
Is it able to define a declarative primitive with multiple outputs synchronized to different clocks?
For example, I wonder the following dual port RAM entity is possible in CLaSH,
```VHDL
entity dual_port_ram is
	port 
	(
		clk_a	: in std_logic;
		clk_b	: in std_logic;
		addr_a	: in natural range 0 to 7; -- synchronized to clk_a
		addr_b	: in natural range 0 to 7; -- synchronized to clk_b
		data_a	: in std_logic_vector(15 downto 0); -- synchronized to clk_a
		data_b	: in std_logic_vector(15 downto 0); -- synchronized to clk_b
		we_a	: in std_logic := '0'; -- synchronized to clk_a
		we_b	: in std_logic := '0'; -- synchronized to clk_b
		q_a	: out std_logic_vector(15 downto 0); -- synchronized to clk_a
		q_b	: out std_logic_vector(15 downto 0) -- synchronized to clk_b
	);
end dual_port_ram;
```

leonschoorl, 2017-08-14:
Yes it can.
The trick is having multiple outputs with different clocks is to output a tuple with Signal synchronized to different clocks.

```haskell
topEntity :: Signal' ClkA Addr
          -> Signal' ClkB Addr
          -> (Signal' ClkA Value, Signal' ClkB Value)
```

Writing a correct Haskell simulation of a dual port RAM might be tricky, but writing a primitive for it is straightforward.
See the tutorial for more about [multiple clock domains](https://hackage.haskell.org/package/clash-prelude-0.11.2/docs/CLaSH-Tutorial.html#g:13) and writing [primitives](https://hackage.haskell.org/package/clash-prelude-0.11.2/docs/CLaSH-Tutorial.html#g:14)

Here is an example which exposes the read functionality of your dual_port_ram entity (making it a dualPortROM).

MultiClockPrim.hs:
```haskell
module MultiClockPrim where

import CLaSH.Prelude
import CLaSH.Prelude.Explicit

type Addr = Unsigned 3
type Value = BitVector 16

dualPortROM :: SClock clka
            -> SClock clkb
            -> Signal' clka Addr
            -> Signal' clkb Addr
            -> (Signal' clka Value, Signal' clkb Value)
dualPortROM clka clkb addrA addrB = (undefined,undefined)
{-# NOINLINE dualPortROM #-}

topEntity = dualPortROM clkA clkB

type ClkA = 'Clk "A"    45
type ClkB = 'Clk "B"    100

clkA :: SClock ClkA
clkA = sclock

clkB :: SClock ClkB
clkB = sclock
```

MultiClockPrim.json
```json
[{ "BlackBox" :
    { "name" : "MultiClockPrim.dualPortROM"
    , "type" :
"dualPortROM :: SClock clka        -- ARG[0]
             -> SClock clkb        -- ARG[1]
             -> Signal' clka Addr  -- ARG[2]
             -> Signal' clkb Addr  -- ARG[3]
             -> (Signal' clka Value, Signal' clkb Value)"
    , "templateD" :
"
~GENSYM[~COMPNAME_dualPortRom][0] : block
  signal ~GENSYM[res_a][1] : std_logic_vector(15 downto 0);
  signal ~GENSYM[res_b][2] : std_logic_vector(15 downto 0);
begin
  ~GENSYM[~COMPNAME_dual_port_ram][3] : entity dual_port_ram
  PORT MAP (
    clk_a      => ~CLK[0],
    clk_b      => ~CLK[1],
    addr_a     => to_integer(~ARG[2]),
    addr_b     => to_integer(~ARG[3]),
    data_a     => (others => '0'),
    data_b     => (others => '0'),
    q_a        => ~SYM[1],
    q_b        => ~SYM[2]
  );
  ~RESULT <= (~SYM[1],~SYM[2]);
end block;"
    }
}]
```

alex-mckenna, 2021-08-13:
Closing: the question was answered and the issue is inactive

closed 2021-08-13 by alex-mckenna