clash-lang/clash-compiler #241: One day, remove lambdaDrop

thoughtpolice, 2017-08-16:
Based on a convo with @christiaanb: provided we can fix [GHC issue #13663](https://ghc.haskell.org/trac/ghc/ticket/13663), we could get rid of the lambdaDrop pass completely. This would be nice (and also sort of nip some of its bad time complexity in the bud).

It might be possible to do this for GHC 8.2.2 if someone writes the patch and asks very nicely. Shouldn't be too hard.

christiaanb, 2017-08-24:
With 65dacdc0cc515737edf1ac8e0ee8343eea881a5d we no longer need lambaDrop because Clash will only get self-recursive bindings, a group of `N` mutually recursive bindings is turned into `N` self-recursive binders.

Nonetheless, it would be nice if [GHC issue #13663](https://ghc.haskell.org/trac/ghc/ticket/13663) is still fixed.

thoughtpolice, 2017-09-03:
Good enough for me -- I'll close this in the mean time, and perhaps open another ticket filled with "GHC Wishlist" stuff.

closed 2017-09-03 by thoughtpolice