clash-lang/clash-compiler #251: Bad compilation performance when creating pipelines via folds

blaxill, 2017-09-04:
I have a design that is a simple pipeline design, where each stage is similar up to a configuration value. I would like to represent this as a fold for brevity and clarity, unfortunately normalisation in that form is prohibitive. Unrolling is a workaround, but the fold is preferred. The issue is similar to #236, but that issue does not provide a solution to the fold/function composition. 

Performance of the first foldr version show below seems like a bug to me, although I don't know how the Clash compiler works. Perhaps there is a better way to write the function composition but without resorting to manual unrolling?   

```Haskell
import CLaSH.Prelude
import Data.Word

topEntity :: Signal Word32 -> Signal Word32
topEntity = pipeline
  where
    pipeline :: Signal Word32 -> Signal Word32
    -- slow normalisation, 32s
    pipeline = foldr (\x acc -> stage x . acc) id $ iterate d32 (+1) 1
    -- slow normalisation, killed after 60s
    pipeline = foldr (.) id $ map stage $ iterate d32 (+1) 1
    -- fast normalisation, 1s
    pipeline =
        stage 1 .  stage 2 .  stage 3 .  stage 4 .  stage 5 .  stage 6 .
        stage 7 .  stage 8 .  stage 9 .  stage 10 .  stage 11 .  stage 12 .
        stage 13 .  stage 14 .  stage 15 .  stage 16 .  stage 17 .  stage 18 .
        stage 19 .  stage 20 .  stage 21 .  stage 22 .  stage 23 .  stage 24 .
        stage 25 .  stage 26 .  stage 27 .  stage 28 .  stage 29 .  stage 30 .
        stage 31 .  stage 32

    -- Simple example where `stage` is configured by its order in the pipeline
    stage :: Word32 -> Signal Word32 -> Signal Word32
    stage i s = let delay = register i $ xor <$> delay <*> s 
                in delay
```

Note I did also try with foldl and plain Haskell versions of the functions, but all suffered from the performance issue.

christiaanb, 2017-09-04:
As a workaround you can write:

```haskell
topEntity i = last stages
  where
    stages = zipWith stage (iterate d32 (+1) 1) (i :> init stages)
```

which compiles much faster.

blaxill, 2017-09-05:
Thanks thats a lot quicker. I'm guessing that this avoids some symbolic computation required in the fold method? I'm interested in contributing back to this project, is there an obvious place I could start looking for applying an optimization to speed up the fold methods? (I'm assuming this is technically possible as the results are equivalent.)  For anyone following the examples, the iterate needs to be flipped `(iterate d32 (subtract 1) 32)`.

christiaanb, 2017-09-07:
Sorry for the late response, was at ICFP. Anyhow, you are correct that in the version I gave we can avoid symbolic evaluation of all the higher-order functions, and instead pick a specialised code-path which converts them to loops in VHDL/Verilog.

Whether Clash is forvced to symbolically evaluate a (higher-order) function or pick a special fast code-path depends on whether the arguments are "representable", i.e. have a straightforward bit-representation. In your example, you are create a vector of functions, so the `fold` cannot be translated to a simple loop over its elements.

So, the Clash compiler has bad asymptotics (and perhaps bad constants) for certain inputs, and it seems to be easily triggered by functions such as the above; which is good to know, because that mean we have a simple test case by which we can measure any fixes for these bad asymtotics.

So to give some pointers:
- https://github.com/clash-lang/clash-compiler/blob/44e647f93655248a211deee6a77ad0aa8518a3f3/clash-lib/src/CLaSH/Normalize/Transformations.hs#L1074 is the transformation which decides whether certain HO-primitives get symbolically evaluated or not.
- https://github.com/clash-lang/clash-compiler/blob/44e647f93655248a211deee6a77ad0aa8518a3f3/clash-lib/src/CLaSH/Normalize/PrimitiveReductions.hs#L289 is the function which fully unrolls `foldr`

However, I don't think it's the actual unrolling of `foldr` that is the problem, but the subsequent normalisation. The entire normalisation procedure is specified in: https://github.com/clash-lang/clash-compiler/blob/44e647f93655248a211deee6a77ad0aa8518a3f3/clash-lib/src/CLaSH/Normalize/Strategy.hs and the individual transformation here: https://github.com/clash-lang/clash-compiler/blob/44e647f93655248a211deee6a77ad0aa8518a3f3/clash-lib/src/CLaSH/Normalize/Transformations.hs

Probably the "easiest" place to start is in the transformations, and see where some information is recomputed instead of being reused, or where "expensive" checks are performed by before "cheap" checks. Maybe some transformations can be merged (see `Strategy.hs` which are logically grouped), so that all their checks happen once, instead of successively. Although these function form the "core" of Clash' functionality, the changes/fixes which I'm proposing above are all "simple" refactorings, and are hopefully not to scary. Also, the regression suite should catch any mistakes.

blaxill, 2018-10-11:
This now compiles a lot faster now due to the move away from `unbound-generics`

~~~Haskell
pipeline = foldr (\x acc -> stage x . acc) id $ iterate d32 (+1) 1
~~~

(The `foldr (.) id $ map stage $ …` appears to still be slow but is not important in my case.)

closed 2018-10-11 by blaxill