clash-lang/clash-compiler #258: Can't install

atg, 2017-09-25:
Followed instructions on the website, as far as I can tell. 

```
$ cabal install clash-ghc --enable-documentation
Resolving dependencies...
cabal: Could not resolve dependencies:
trying: clash-ghc-0.7.2 (user goal)
trying: base-4.10.0.0/installed-4.1... (dependency of clash-ghc-0.7.2)
next goal: ghci (dependency of clash-ghc-0.7.2)
rejecting: ghci-8.2.1/installed-8.2..., ghci-8.2.1 (conflict: clash-ghc =>
ghci>=8.0.1 && <8.2)
rejecting: ghci-8.0.2, ghci-8.0.1 (conflict: base==4.10.0.0/installed-4.1...,
ghci => base==4.9.*)
After searching the rest of the dependency tree exhaustively, these were the
goals I've had most trouble fulfilling: clash-ghc, template-haskell,
clash-lib, lens, clash-prelude, clash-verilog, base, ghc, reflection, unbound,
RepLib, mtl, ghci
```

```
$ cabal --version
cabal-install version 2.0.0.0
compiled using version 2.0.0.2 of the Cabal library 

$ ghc --version
The Glorious Glasgow Haskell Compilation System, version 8.2.1
```

macOS 10.12.4

christiaanb, 2017-09-25:
Yeah, there's a documentation error on the website, Clash only compiles with version 8.0.* right now, not higher or lower. I'll see if I can release the new version of Clash tonight which is one that works on GHC 8.2.*.

leonschoorl, 2017-09-27:
Maybe we should change the installation instructions to use stack?

1. Install Stack, by following [these instructions](https://docs.haskellstack.org/en/stable/README/#how-to-install).
2. Use stack to install Clash: `stack install --install-ghc --haddock clash-ghc`
This should automatically install the correct GHC version, Clash and all its dependencies.

The downside to this is that you now should run clash via stack.
Like this: `stack exec -- clash --vhdl FIR.hs`

@atg you could try this, it should get you GHC-8.0.2 and clash-0.7.2

silkeh, 2018-02-08:
When installing with Stack, note that the latest LTS release contains GHC-8.2, so the LTS-9 release has to be specified manually:
```
stack install --install-ghc --haddock clash-ghc --resolver lts-9
```

christiaanb, 2018-07-26:
Installation issue should be fixed now, the documentation is now in line with what's actually possible: Clash can only install with GHC 8.2 and GHC 8.4

closed 2018-07-26 by christiaanb