clash-lang/clash-compiler #266: Bug in shift for verilog and shift missing.

threonorm, 2017-11-11:
Hi,

In the verilog backend for GHC primitive,  it seems that in this case left becomes right:

```

 263    { "BlackBox" :
 264     { "name"      : "GHC.Prim.uncheckedIShiftL#"
 265     , "type"      : "uncheckedShiftRL# :: Word# -> Int# -> Word#"
 266     , "templateE" : "~ARG[0] >> ~ARG[1]"
 267     }
 268   }

```

where the vhdl seems ok:

```

 157 , { "BlackBox" :
 158     { "name"      : "GHC.Prim.uncheckedIShiftL#"
 159     , "type"      : "uncheckedIShiftL# :: Int# -> Int# -> Int#"
 160     , "templateE" : "shift_left(~ARG[0],to_integer(~ARG[1]))"
 161     }
 162   }

```

Should I submit a pull request? I also noticed that at least on shift  (uncheckedShiftRL) is not present in the verilog translation when it is present in the vhdl translation that.

threonorm, 2017-11-11:
(Also, I am not sure I understand why sometimes unchecked integer shift are translated to unchecked shift in the types for the templates of verilog. That seems suspicious but maybe I am missing something.)

ggreif, 2017-11-11:
I think it would be great to submit a pull request (unless `master` is fixed already). Do not forget to add a test if you decide to do so.

leonschoorl, 2017-11-13:
Indeed, it is named incorrectly.
The type is correct and name should match that. (type fields in the json files are just comments, that are (currently) not being parsed by the compiler)

A pull request would be great. If you do, please also fix the sytemverilog primitives, they have exactly the same problem.

closed 2018-01-16 by christiaanb