clash-lang/clash-compiler #280: Surprising type defaulting

leonschoorl, 2017-12-07:
Taking the 3 least significant bits of 5 in clashi:
```
Clash.Prelude> slice d2 d0 5
000
```

Why not `101`?

Because of [type defaulting](https://downloads.haskell.org/~ghc/latest/docs/html/users_guide/ghci.html#type-defaulting-in-ghci), clashi picks the type of 5 to be Double:
```
Clash.Prelude> slice d2 d0 (5::_)
<interactive>:11:17: error:
    • Found type wildcard ‘_’ standing for ‘Double’
```

Note that this only happens in clashi.
Doing the same in a .hs file, results in an ambiguous type variable error.

But I think clashi should never default to Double.

masterdezign, 2018-04-05:
Hi!

I think, there is no reason for `101` unless bu default `5 :: Unsigned 32` or something like that. 
If by default `5 :: Double`, than it means that `slice d2 d0` takes the tail of the fractional part, which is zero. That is why you get `000`.

christiaanb, 2018-12-18:
We should get an ambiguity error

PR #1021 (2020-01-17): Add `Int` before `Double` in interactive defaulting rules
Fixes #280
commit fa52132f (2020-01-17): Add `Int` before `Double` in interactive defaulting rules

Fixes #280
closed 2020-01-17 by christiaanb