clash-lang/clash-compiler #284: Varying Initial State

kleinreact, 2017-12-18:
It seems that for some code constucts the generated components are not properly initialized. I tried the following code on an iceStick40 board. The code lets one of the board leds blink.
```haskell
module TopEntity where

import Clash.Prelude

data Mode = On | Off

{-# ANN topEntity (
  defTop
    { t_name    = "Test"
    , t_inputs  =
        [ PortField ""
            [ PortName "CLOCK"
            , PortName "RESET"
            ]
        ]
    , t_output =
        PortName "LED"
    }
  )
  #-}

topEntity
  :: SystemClockReset
  => Signal System Bit

topEntity =
    stateflip $ prescale (12000000 :: Unsigned (CLog 2 12000000))
  
{-# NOINLINE topEntity #-}
  

stateflip
  :: SystemClockReset
  => Signal System Bit -> Signal System Bit

stateflip = mealy flipBit (Off, low)
  where
    flipBit state ps =
      let
        newstate@(_, output)
          | ps == low  = state
          | otherwise = nextState state
      in
        (newstate, output)

    -- works
    nextState (mode, output) = case mode of
      On               -> (Off, low)
      Off              -> (On, high)        

    -- does not work
    nextState' (mode, output) = case mode of
      On
        | output == low -> (mode, output)
        | otherwise    -> (Off, low)
      Off              -> (On, high)

prescale
  :: (SystemClockReset, KnownNat n)
  => Unsigned n -> Signal System Bit

prescale x = mealy transition (x - 1) $ pure ()
  where
    transition y ()
      | y > 0     = (y - 1, low)
      | otherwise = (x - 1, high)    
```
If I compile it to verilog using the `nextState` function everything works fine and the led starts blinking immediately after the upload. If I however use `nextState'`, the led is turned off initially and only starts blinking after resetting the system via the reset button.

In the simulation everything works fine for both versions. Hence I assume some code generation related bug.

I used the latest available clash versions:
- clash-compiler: commit bd87c48
- clash-prelude: commit db6b8a1

christiaanb, 2017-12-18:
By simulation, you mean simulation in Clash? or simulation in Verilog?

kleinreact, 2017-12-18:
I mean simulation in Clash. I haven't tried simulation of the Verilog code, but I compiled it with yosys and the lattice toolchain, both showing the same difference.

christiaanb, 2017-12-18:
Ah, so the `simulation` function does an implicit reset.

So what's happening in your circuit is the following:

1. `On` is encoded as `0` and `Off` is encoded as `1`.
2. The registers probably power up in the `0` state
3. Looking at `nextState'`, on power up, you're in the `On` state and the `output` is `low`, and you'll be stuck in that state forever until you explicitly reset.

So, probably, if you switch the order of the constructors in the data declaration of `Mode` to:

```haskell
data Mode = Off | On
```

You probably won't see the difference between `nextState` and `nextState'`, as it powers up in the `Off` state.

kleinreact, 2017-12-18:
You are right, switching the order indeed solves the problem in case of `nextState'`. 

So do I understand it right that the initial state passed to the `mealy` function is only used for the reset? Is this intended behavior? From the description I would expect that it also generates some
```verilog
initial begin
...
end
```
verilog code initially assigning the passed initial value to the state of the state machine.

christiaanb, 2017-12-18:
So I guess you can say in general a reset is used to bring a circuit into a predefined state; which from the circuits point of view is its initial state. So yes, the mealy machine needing a `reset` to get to its initial state is very much intended, but perhaps not obvious from the documentation.

Also, while `initial` blocks might work for FPGAs because they can be used synthesized to the values of the configuration SRAMs; `initial` block cannot used for ASICs, because the power-up values of registers are basically undefined.

kleinreact, 2017-12-18:
I agree with your point. 

However, the difference then is that with Verilog I have at least the possiblity to assign a defined initial state, while with clash I have not (if I understood you correctly). In the end, nobody really wants to have a system starting in some undefined state, where the only exception might be if you have some registers that don't need to be initialized since you know they will be assigned the right value after some steps anyway. But I would see those situations more as the special case, than as the default.

My suggestion would be that there is at least some option to select whether registers are also "reset" on power up or not. For those who do not need this, they then simply can disable this option. 

I would prefer proper initialization as the default, as the outcome then coincides with the clash simulation and, hence, with the expected behavior. The point is, that if you don't think about this problem in the first place, it may get you some to long debugging sessions. Especially, as many things can just work right by accident, while just some tiny bit keeps to be wrong.

christiaanb, 2017-12-18:
So the problem is, for ASIC, the only way to get your registers in a defined state is through a reset. ASIC synthesis tools completely ignore `initial` blocks in Verilog.

I guess I could have Clash create `initial` blocks for registers; in principal it doesn't hurt anyone if it did.

martijnbastiaan, 2019-02-15:
PR #498 implements register power up values, which solves this issue. Thanks for the report!

closed 2019-02-15 by martijnbastiaan