clash-lang/clash-compiler #285: Clashi code generation fails in multi-file setups

sonOfRa, 2017-12-21:
Environment: 
- Gentoo Linux, 64-bit
- GHC 8.0.2
- CLaSH 0.7.2

When building a multi-file setup, like https://github.com/clash-lang/clash-compiler/blob/v0.7.2/examples/Calculator.hs in clashi, VHDL generation fails with the following error:

```
ByteCodeLink.lookupIE
During interactive linking, GHCi couldn't find the following symbol:
  CalculatorTypes_Imm_con_info or CalculatorTypes_Imm_static_info
```

The error does not occur when invoking ```clash --vhdl Calculator.hs``` directly, in which case proper VHDL code is generated.

Compilation of the code itself works, as clashi seems to successfully load the files, and only when I try to generate VHDL, the generation fails.

I ran across the error when building some custom types for a university course, code and error are [here](https://gist.github.com/sonOfRa/5545fc3d5ef522f6b86f663bbb66c10e). Since the error does NOT occur when I use the literal values in line 36 in Memory.hs in my above example, it looks to me like it might be a problem with how template-haskell is handled in ghci. However that is speculation from my side as I do not really know a lot about Haskell in general or its internals, especially not template-haskell

nlhans, 2018-02-01:
I also encountered this issue, and want to contribute to this ticket by providing a second, bare bones, code example. I'm confident the issue is directly related to the listToVecTH function:

https://gist.github.com/nlhans/5a7a8f5c49c3be13ab28055aac078fc4

Interacting with the code in clashi works fine, as I'm able to call topEntity (1:>1:>Nil) and receive the expected value. 
However when compiling to vhdl inside clashi, I get the same error. But I was able to generate the code by calling clash instead.

Environment:

- Manjaro Linux 17.1.1 
- GHC 8.0.2
- CLaSH 0.7.2, clash-lib 0.7.1
- Running this project via stack, as described in #258

christiaanb, 2020-01-13:
Fixed in `master`

closed 2020-01-13 by christiaanb