clash-lang/clash-compiler #287: Mismatch between interactive evaluation and compilation or loading

cemerick, 2018-01-10:
Given a file `Foo.hs`:

```hs
module Foo where

import Data.Int

x = 56 :: Int8
jj :: Int8 -- whether this is here or not doesn't eliminate the error
jj = x + x
```

…either compilation _or_ loading into `clash --interactive` (via `:l Foo`) yields:

```
src\Foo.hs:6:8: error:
    Variable not in scope: (+) :: Int8 -> Int8 -> t
```

However, doing the same work interactively in `clash --interactive` does what I'd expect:

```hs
CLaSH.Prelude> import Data.Int
CLaSH.Prelude Data.Int> let x = 56 :: Int8
CLaSH.Prelude Data.Int> x + x
112
```

I am using v0.7.2 at the moment. Any ideas?

(In case it isn't clear, I'm brand new to clash, and it's my first serious foray in Haskell in ~10 years. 🙈  Thank you in advance for any assistance you might offer. ❤️ )

christiaanb, 2018-01-10:
Unlike "normal" Haskell compilers, `clash` doesn't import an implicit prelude, because it has two versions:
* One that has implicit/hidden clocks: `CLaSH.Prelude`
* One that has explicit clocks: `CLaSH.Prelude.Explicit` (will be renamed in 0.99/1.0 to: `Clash.Explicit.Prelude`)

So adding:

```haskell
import CLaSH.Prelude
```

to your `Foo.hs` file should fix things.

cemerick, 2018-01-10:
Ah, so I could have inferred that things were working interactively because of the `CLaSH.Prelude` showing up in the prompt.

Thank you very much!

closed 2018-01-10 by cemerick