clash-lang/clash-compiler #295: maxBound in Index n creates 64bit signed verilog value regardless of n, causing usage in tuples to compose wrong value

ejelly, 2018-02-11:
This very simple code:

```
topEntity :: Signal Bool -> Signal ((Index 4, Index 4))
topEntity _ = bundle $ (signal maxBound, signal maxBound)
```

generates the following verilog code in `*_topEntity_0.v`:

```
  output [3:0] result;
  assign result = {64'sd4-2'd1
                  ,64'sd4-2'd1};
```

The expression `64'sd4` in the concatenation is of the wrong size (and signedness?), and causes the `result`, truncated to the correct length, to clobber over the upper tuple values (making them all 0 unless you have really big tuples). Consequently, the code simulates fine in clash, but fails simulation in Vivado.

As workarounds, using the actual maximum integer literal instead of `maxBound` or using e.g. `Unsigned 2` instead both produces the correct verilog expression (`2'd3` or `{2 {1'b1}` respectively), but this limits available polymorphism.

`CLaSHi, version 0.7.0.1 (using clash-lib, version 0.7)`

ejelly, 2018-02-11:
Seeing that #79 reported almost the same issue, I wonder if the fix is just applying the same change as https://github.com/clash-lang/clash-compiler/pull/79/files#diff-5a7a995011ffcca15d76c14bbf171ed3 to the verilog template as well?

christiaanb, 2018-02-12:
So the source of the bug is that Clash renders `Integer`, which is the underlying representation for `KnownNat n` as 64-bit signed integers in Verilog; where in this case we would actually just want the literal.

Changing the Primitive definition to:

```
[ { "BlackBox" :
    { "name"      : "CLaSH.Sized.Internal.Index.maxBound#"
    , "type"      : "maxBound# :: KnownNat n => Index n"
    , "templateE" : "~LIT[0]-~SIZE[~TYPO]'d1"
    }
  }
]
```

should work I think.

In case you want to try you can make a `*.json` file with the above content in the working directory and the clash compiler will overwrite the bundled primitive.

ejelly, 2018-02-12:
Thanks, but apparently that's not enough. The generated code is:
```
  assign result = {4-2'd1
                  ,4-2'd1};
```
and while this looks better, simulation of my simple test case still does not pass. I guess this still uses the 32bit default width in verilog.

Here's my full, simple test case by the way:

```
topEntity :: Signal Bool -> Signal ((Index 4, Index 4))
topEntity _ = bundle $ (signal maxBound, signal maxBound)

testInput :: Signal Bool
testInput = signal True
expectedOutput = outputVerifier (((3, 3) :: (Index 4, Index 4)) :> Nil)
```

christiaanb, 2018-02-12:
I guess until we have constant folding inside templates, the only solution for now then seems to be:

```
[ { "BlackBox" :
    { "name"      : "CLaSH.Sized.Internal.Index.maxBound#"
    , "type"      : "maxBound# :: KnownNat n => Index n"
    , "templateD" : "assign ~RESULT = $unsigned(~LIT[0]-~SIZE[~TYPO]'d1)"
    }
  }
]
```

where the verilog assignment will take care of the resize.

christiaanb, 2018-07-26:
This seems fixed in the latest released version, I'm now getting:

```verilog
assign result = {2'd3,2'd3};
```

closed 2018-07-26 by christiaanb