clash-lang/clash-compiler #299: ~FILE[...] doesn't work inside ~IF ~THEN ~ELSE

basile-henry, 2018-03-06:
When writing a custom Clash primitive, the template language doesn't seem to resolve filepaths inside if-then-else constructs.

This is the error message I get:

```
    <no location info>: error:        
        Clash error call:             
        Clash.Netlist.BlackBox.Util(425): argument of ~FILEPATH:"~LIT[6]"does not reduce to a string
```

A workaround I found was to assign `~FILE[~LIT[6]]` to a VHDL `constant` and then use the constant in the if-then-else.

By the way, I am using the [Clash Tutorial page](https://hackage.haskell.org/package/clash-prelude-0.11.2/docs/CLaSH-Tutorial.html#g:14) as a reference for the template language and it seems to be missing the `~FILE` construct!

martijnbastiaan, 2018-09-26:
I'm confused, do you want to use `~FILE[~LIT[6]]` as a conditional value? Could you provide an example maybe?

basile-henry, 2018-09-26:
The use case was to only use the `~FILE` construct depending on the filename. Intel/Quartus seems to be okay with an empty string to leave memory uninitialised whereas Xilinx/Vivado needs the string `"none"` for the filepath.

I found a workaround which goes as follows (in a `VHDL` template):

```json
"type":
"
  ...
  -> FilePath                                                        -- ARG[6]
  ..."
...
"template":
"
  ...
  constant ~GENSYM[filepath][2] : string := ~FILE[~LIT[6]];
  ...
      -- string; \"none\" or \"<filename>.mem\"
      MEMORY_INIT_FILE => ~IF~STRCMP[none][6]~THEN\"none\"~ELSE~SYM[2]~FI,
  ..."
```

When I wasn't using the `constant` and just used `~FILE[~LIT[6]]` inside the `~IF ... ~THEN ... ~ELSE ... ~FI` the template failed to parse properly.

The error message at the time was:
```
argument of ~FILEPATH:"~LIT[6]"does not reduce to a string
```

martijnbastiaan, 2018-09-26:
Got it, thanks, I'll have a look!

christiaanb, 2020-01-13:
This works on `master`

closed 2020-01-13 by christiaanb