clash-lang/clash-compiler #300: Recursively, but terminating, contant not inlined

christiaanb, 2018-03-07:
The following code results in a Clash synthesis error:

```haskell
{-# LANGUAGE GADTs #-}
module Test where

import Clash.Prelude hiding (fold)
import Data.Foldable (fold)

data SomeSNat where
  SomeSNat :: forall n . SNat n -> SomeSNat

instance Monoid SomeSNat where
  mempty = SomeSNat d0
  (SomeSNat a) `mappend` (SomeSNat b) = SomeSNat (a `addSNat` b)

a :: SomeSNat
a = SomeSNat d8

b :: SomeSNat
b =
  fold
    [ a
    , SomeSNat d2
    ]

topEntity
  :: Integer
  -> Integer
topEntity x = case b of
  SomeSNat s -> snatToInteger s + x
```

where the problem is that Clash refuses to inline the `fold`. 

The reason that Clash doesn't inline `fold` is because:
1. it is recursive and 
2. has a representable type

This is a problem, because Clash has two inlining mechanism:

1. `inlineNonRep` which inlines everything that has a non-representable type, even if it is recursive. It has a simple termination measure by which it checks that it only inlines a value `f` into `g` at most `X` times.
2. `inlineWorkFree` which inlines everything that doesn't add to the size of the circuit, except when those things are recursive. The reason it doesn't inline recursively defined work-free things is because there are no termination measures in place.

Likely solution: add termination measure for `inlineWorkFree` similar to `inlineNonRep` for recursively defined values.

basile-henry, 2018-03-07:
Is the `Integer` part not an issue? Since it doesn't have a bounded representation?

I think the issue is still there if you use `Int` for example, I'm just wondering if `Integer` doesn't add more issues in this specific example.

christiaanb, 2018-03-07:
`Integer` is fine because it's interpreted incorrectly as a 64-bit signed.

alex-mckenna, 2021-08-13:
Closing: this issue no longer occurs with the given example

closed 2021-08-13 by alex-mckenna