clash-lang/clash-compiler #304: Derived numeric constant causes compiler hang/failure

cemerick, 2018-03-20:
I have a numeric constant:

```hs
a = r - Foo.c
```

This `a` is used in a function in the same module. The compiler fails/hangs after outputting `Compiling: Core.Bar.topEntity` (doesn't return after an hour waiting).

(`Foo.c` is a `Signed 520`, FWIW.)

Adding an inline pragma for `a` doesn't help, but replacing the expression with a numeric literal does, e.g.

```hs
a = 234576238764
```

With this change, the compilation for this particular circuit succeeds in ~20s.

Hopefully the above provides enough clues to track down the problem. If not, I could probably produce a minimal example.

leonschoorl, 2018-03-29:
I tried reproducing this, but failed.

Could you provide some more information, or preferably an example?

cemerick, 2018-03-30:
Here's as minimal a case as I can manage:

```hs
module Core.Foo where

import Clash.Prelude

type IntT = Signed 520

m = 92384792359876348747637737 :: IntT
r = 2 ^ 192

{-# INLINE a #-}
a = r - m

data Step = S1 | S2

fooT step inp =
  case step of
    S1 -> (step, inp * a)

topEntity :: SystemClockReset =>
             Signal System IntT ->
             Signal System IntT
topEntity = mealy fooT S1
```

To clarify the way in which the compilation fails, the compiler is actually churning through memory. It gets up to ~2.5G resident before the process is killed.

There's actually _three_ things that will get this to compile (discovered while whittling down the real code to a minimal example):

1. Replace `r - m` with a numeric literal, as mentioned previously.
2. Remove `S2`.
3. Lift the usage of `a` out of the `case` expression, i.e. so `fooT` reads like so:

```hs
fooT step inp = (step, inp * a)
```

christiaanb, 2018-03-30:
Replacing `r` by:

```
r = 6277101735386680763835789423207666416102355444464034512896 -- 2 ^ 192
```

compiles.
The original error might be due to `^` horrible recursive definition: http://hackage.haskell.org/package/base-4.11.0.0/docs/src/GHC.Real.html#local-6989586621679040551

Investigating further

cemerick, 2018-03-30:
That's interesting, I guess I've been assuming that expressions involving constants like that would get flattened by ghc before clash saw them. (But then I haven't yet actually read any of ghc, alas.)

I guess I'll just use `shiftL` instead, so as to retain clarity about what the constant is, etc.

commit 836c6d36 (2018-03-30): Evaluator: rename let-binders onto the heap

Otherwise seperate calls to a function `f` that has
let-binders, would overwrite each others heap-bound
values.

Fixes #304
closed 2018-03-30 by christiaanb