clash-lang/clash-compiler #311: Bug with lifting of initial blockRam contents

wyager, 2018-04-15:
Some caveats:

* I'm not on 0.99. For some reason cabal doesn't want to recognize that it exists. I'm on clash-ghc-0.7.2. Maybe this was fixed since then
* I'm not even sure if this is a Clash bug or an iverilog bug, although I have used iverilog 10.1 before with no issues. The issue also persists on iverilog 11.0

OK, here's the skinny.  Check out this code:

```haskell
module Test where

import CLaSH.Sized.Unsigned (Unsigned)
import CLaSH.Sized.Vector (Vec((:>), Nil))
import CLaSH.Sized.Index (Index)
import CLaSH.Prelude.BlockRam (blockRam)
import CLaSH.Signal (signal, register)
import CLaSH.Prelude 

topEntity = go $ f $ ((1 :: Unsigned 8) :> 3 :> 6 :> 7 :> Nil) 
    where
    -- f = fmap (+1) -- Undefined
    -- f = fmap (\x -> x) -- High impedance
    f = (\x -> x) -- OK
    go vec = ram
        where
        ram = blockRam vec read (signal Nothing)
        read = register (0 :: Unsigned 2) (1 + read)
```

Depending on what definition of `f` you choose, you get a different result when running the synthesized hardware in `iverilog`. I believe the problem is related to `blockRam` where the initial contents are non-literal.

Here's the verilog code:
```verilog
`timescale 1ns/1ns

module main(clk, reset);
    input clk;
    input reset;
    initial begin
        $dumpfile("test.vcd");
        $dumpvars(0,cpu);
    end

    // Toggle the reset line
    initial begin
        reset_reg = 1;
        reset_reg = #1 0;
        reset_reg = #2 1; 
        counter = 0;
    end
    reg reset_reg;
    wire reset = reset_reg;
    
    // Clock line
    reg theClock = 0;
    assign clk = theClock;
    always begin
        #50;
        theClock = !theClock;
    end

    wire [7:0] output_data;

    Test_topEntity cpu(clk, reset, output_data);
    
    reg [2:0] counter;

    always@(posedge clk) begin
        counter = counter + 1;
        $display("0x%h", output_data);
        if (counter == 4'b111) begin
            $finish;
        end
    end
endmodule
```

If `f = id`, you get 

```
0xxx
0x01
0x03
0x06
0x07
0x01
0x03
```

(correct)

`Test_topEntity_0.v` has

```verilog
  reg [31:0] ram_init;
  integer i;
  initial begin
    ram_init = {8'd1,8'd3,8'd6,8'd7};
    for (i=0; i < 4; i = i + 1) begin
      RAM[4-1-i] = ram_init[i*8+:8];
    end
  end
```

If `f = fmap id`, you get

```
0xxx
0xzz
0xzz
0xzz
0xzz
0xzz
0xzz
```

Code looks like

```verilog
  wire [31:0] app_arg;
...
  assign vec = {8'd1,8'd3,8'd6,8'd7};
  
  genvar i;
  generate
  for (i=0; i < 4; i = i + 1) begin : map
    wire [7:0] map_in;
    wire [7:0] map_out;
  
    assign map_in = vec[i*8+:8];
    Test_topEntity_specF Test_topEntity_specF_0
  (.result (map_out)
  ,.x (map_in));
    assign app_arg[i*8+:8] = map_out;
  end
  endgenerate
...
 reg [31:0] ram_init;
  integer i_0;
  initial begin
    ram_init = app_arg;
    for (i_0=0; i_0 < 4; i_0 = i_0 + 1) begin
      RAM[4-1-i_0] = ram_init[i_0*8+:8];
    end
  end
```

If `f = fmap (+1)`, you get

```
0xxx
0xxx
0xxx
0xxx
0xxx
0xxx
0xxx
```

Same deal as above.

I initially ran into this issue when the RAM contents were being passed in to the module through a wire. 

To build:
```sh
clash --verilog Test.hs
iverilog -o test -s main test.v verilog/Test/*.v
./test
```

This is definitely a new problem for me - I've used `blockRam` in the past with no issues.

christiaanb, 2018-04-15:
So the bug/issue is that Clash tries to avoid evaluating "expensive" functions (like `Clash.Sized.Vector.map`) at compile-time unless it's required to be able to produce Verilog. What's going on here is that Clash doesn't really understand that the initial content of the BlockRAM must be a literal in order for the Verilog to behave the same as the Haskell code. I'll try and fix it.

Some notes:
* It's broken on `master` as well, not just 0.7.2
* Are you on GHC 8.2.1 or higher? if so what's the error you get when forcefully trying to install `cabal install --force-reinstalls clash-ghc-0.99 clash-prelude-0.99 --dry-run` ?

martijnbastiaan, 2019-02-15:
PR #498 adds the ability for blackboxes to request a constant value. If specified, Clash will try extra hard to evaluate functions, even in cases where it would consider it "too expensive" to do previously. This PR is merged and will be part of Clash 1.0, so I'm closing this issue.

closed 2019-02-15 by martijnbastiaan

wyager, 2019-02-15:
Awesome, thanks! I will try to compile my project with this in the near future and see if it works.

martijnbastiaan, 2019-02-15:
Perfect! Let us know how it turns out. You can use the latest Clash by running:

```bash
# Install GHC / cabal / Clash build dependencies
sudo add-apt-repository -uy ppa:hvr/ghc
sudo apt-get install cabal-install-2.2 ghc-8.4.3 libtinfo-dev

# "Install" Clash
git clone https://github.com/clash-lang/clash-compiler.git
cd clash-compiler
cabal new-update
cabal new-run -- clash Test.hs --vhdl
```