clash-lang/clash-compiler #314: Hard-coded -dynamic flag breaks build on openSUSE Linux

peti, 2018-05-02:
I just had to [patch the build of `clash-ghc` for openSUSE Linux/x86-64 with GHC 8.4.2](https://github.com/opensuse-haskell/configuration/commit/8843294a252be6e912d51492c709f6399205bd95), because the hard-coded use of `-dynamic` in the Cabal file broke our post-processing checks:

~~~~
[  412s] calling /usr/lib/rpm/brp-suse.d/brp-35-rpath
[  412s] ERROR: RPATH "/home/abuild/rpmbuild/BUILD/clash-ghc-0.99/dist/build" on /home/abuild/rpmbuild/BUILDROOT/clash-ghc-0.99-2.1.x86_64/usr/bin/clash is not allowed
[  413s] ERROR: RPATH "/home/abuild/rpmbuild/BUILD/clash-ghc-0.99/dist/build" on /home/abuild/rpmbuild/BUILDROOT/clash-ghc-0.99-2.1.x86_64/usr/bin/clashi is not allowed
~~~~

What happens is that `ghc` makes bad decisions about setting certain RPATH entries in the generated executables. It ought to be *Cabal* making those decisions: setting the compiler flag `-dynamic` is not the proper way to enable dynamic linking.

Is there a particular reason why this flag is hard-coded in the Cabal file?

christiaanb, 2018-05-02:
The issue is that if the executables are not dynamically linked we incur a startup time of around 3 seconds.

christiaanb, 2018-05-02:
I'll remove the hard-coded choice in the Cabal flag, and inform users who build Clash from source/cabal should use the `--enable-executable-dynamic` flag. But how would I inform distributors to build Clash using dynamic linking? Is it even possible to have dynamically linked Haskell executables in OpenSUSE taking the RPATH-mentioned issue into account?

peti, 2018-05-02:
Yes, dynamically linked binaries work fine on openSUSE if it's enabled through Cabal.

If you point out the performance gain in the readme or in the build instructions, then I'm sure people who distribute this package will notice it. And even if they don't, then their users will eventually. :-)

Incidentally, many distributions (like openSUSE and NixOS) disable dynamically linked binaries by default because these binaries tend to start up much slower that statically linked ones. I am surprised that it appears to be the other way round with `clash`.

commit a519d1d9 (2018-05-18): Make Cabal do dynamic linking

Fixes #314
closed 2018-05-18 by christiaanb