clash-lang/clash-compiler #315: Installing clash error

YannisMahiou, 2018-05-11:
Hi everyone,
I'm trying to installing Clash on an Ubuntu linux distrib and have some errors that I can't sort out.

cabal install clash-ghc --enable-documentation

```
Resolving dependencies...
Downloading free-5.0.2...
Configuring ghc-typelits-knownnat-0.4.2...
Configuring free-5.0.2...
Building ghc-typelits-knownnat-0.4.2...
Building free-5.0.2...
Failed to install ghc-typelits-knownnat-0.4.2
Build log ( /users/guest/ym4/.cabal/logs/ghc-8.4.1/ghc-typelits-knownnat-0.4.2-FZkfQyT6HMa1cXOwPju162.log ):
cabal: Entering directory '/tmp/cabal-tmp-349/ghc-typelits-knownnat-0.4.2'
Configuring ghc-typelits-knownnat-0.4.2...
Preprocessing library for ghc-typelits-knownnat-0.4.2..
Building library for ghc-typelits-knownnat-0.4.2..
[1 of 3] Compiling GHC.TypeLits.KnownNat.Solver ( src/GHC/TypeLits/KnownNat/Solver.hs, dist/build/GHC/TypeLits/KnownNat/Solver.o )

src/GHC/TypeLits/KnownNat/Solver.hs:261:52: error:
    • Couldn't match type ‘(TcType.TcTyVar, TcType.TcType)’
                     with ‘Var.Var’
      Expected type: [(TcType.TcTyVar, TcType.TcType)]
        Actual type: [((TcType.TcTyVar, TcType.TcType), Ct)]
    • In the first argument of ‘substType’, namely ‘subst’
      In the expression: substType subst z
      In the expression: (x, y, substType subst z)
    |
261 |       kn_wanteds = map (\(x,y,z) -> (x,y,substType subst z))
    |                                                    ^^^^^
cabal: Leaving directory '/tmp/cabal-tmp-349/ghc-typelits-knownnat-0.4.2'
Downloading reducers-3.12.2...
Configuring reducers-3.12.2...
Building reducers-3.12.2...
Installed reducers-3.12.2
Installed free-5.0.2
Updating documentation index
/users/guest/ym4/.cabal/share/doc/x86_64-linux-ghc-8.4.1/index.html
cabal: Error: some packages failed to install:
clash-ghc-0.99-5riqpMAmBN66jz3jDlWKcK depends on clash-ghc-0.99 which failed
to install.
clash-lib-0.99-9959IlDqTBZ94tUYqaQC1x depends on clash-lib-0.99 which failed
to install.
clash-prelude-0.99-7BecyAM3SZGGiF4K5DvIe depends on clash-prelude-0.99 which
failed to install.
ghc-typelits-extra-0.2.4-LK27QSfFS0s2ZJTFBYqkrw depends on
ghc-typelits-extra-0.2.4 which failed to install.
ghc-typelits-knownnat-0.4.2-FZkfQyT6HMa1cXOwPju162 failed during the building
phase. The exception was:
ExitFailure 1
```

I installed both cabal and ghc using the ghc ppa for non-eol distribs.

GHC version :
```
ghc --version
The Glorious Glasgow Haskell Compilation System, version 8.4.1
```

Cabal Version :
```
cabal --version
cabal-install version 2.2.0.0
compiled using version 2.2.0.1 of the Cabal library 
```

Thanks a lot.

christiaanb, 2018-05-11:
Thanks for bringing it to my attention. I know what’s going and I’ll release and updated version of clash within the next few hours. On a airplane now, but will start on a fix once I land. Thanks!

YannisMahiou, 2018-05-11:
Thank you for your responsiveness to my issue.

Have a good flight.

christiaanb, 2018-05-11:
@YannisMahiou I've just pushed what I hope is a fix to hackage. Could you try:

```
cabal update
cabal install clash-ghc --enable-documentation
```

and see if the installation now succeeds?

YannisMahiou, 2018-05-12:
The installation now succeds. Thank you.

chromezh, 2018-05-14:
Hi everyone,

i am sorry to say that my installation on Windows 10 also failed.

```
> cabal install clash-ghc --enable-documentation
Resolving dependencies...
cabal.exe: Could not resolve dependencies:
[__0] trying: clash-ghc-0.99.1 (user goal)
[__1] next goal: process (dependency of clash-ghc)
[__1] rejecting: process-1.6.3.0/installed-1.6... (conflict: clash-ghc =>
Win32>=2.3.1 && <2.6, process => Win32==2.6.1.0/installed-2.6...)
[__1] trying: process-1.6.3.0
[__2] next goal: ghc (dependency of clash-ghc)
[__2] rejecting: ghc-8.4.2/installed-8.4... (conflict: process==1.6.3.0, ghc
=> process==1.6.3.0/installed-1.6...)
[__2] trying: ghc-8.4.1
[__3] next goal: base (dependency of clash-ghc)
[__3] rejecting: base-4.11.1.0/installed-4.1... (conflict: ghc => base<0 &&
==4.11.*)
[__3] rejecting: base-4.11.1.0, base-4.11.0.0, base-4.10.1.0, base-4.10.0.0,
base-4.9.1.0, base-4.9.0.0, base-4.8.2.0, base-4.8.1.0, base-4.8.0.0,
base-4.7.0.2, base-4.7.0.1, base-4.7.0.0, base-4.6.0.1, base-4.6.0.0,
base-4.5.1.0, base-4.5.0.0, base-4.4.1.0, base-4.4.0.0, base-4.3.1.0,
base-4.3.0.0, base-4.2.0.2, base-4.2.0.1, base-4.2.0.0, base-4.1.0.0,
base-4.0.0.0, base-3.0.3.2, base-3.0.3.1 (constraint from non-upgradeable
package requires installed instance)
After searching the rest of the dependency tree exhaustively, these were the
goals I've had most trouble fulfilling: base, clash-ghc, process, ghc,
unbound, RepLib
```

```
> ghc --version
The Glorious Glasgow Haskell Compilation System, version 8.4.2
```

```
> cabal --version
cabal-install version 2.2.0.0
compiled using version 2.2.0.1 of the Cabal library
```

Thanks a lot.

christiaanb, 2018-05-15:
@chromezh I’ve relaxed the dependencies, could you try:

```
cabal update
cabal install clash-ghc —enable-documentation
```

And see if everything works now?

chromezh, 2018-05-16:
I encountered another error:

```
Preprocessing library for clash-ghc-0.99.1..
Building library for clash-ghc-0.99.1..
[ 1 of 13] Compiling Clash.GHC.ClashFlags ( src-ghc\Clash\GHC\ClashFlags.hs, dist\build\Clash\GHC\ClashFlags.o )
[ 2 of 13] Compiling Clash.GHC.GHC2Core ( src-ghc\Clash\GHC\GHC2Core.hs, dist\build\Clash\GHC\GHC2Core.o )
[ 3 of 13] Compiling Clash.GHC.Evaluator ( src-ghc\Clash\GHC\Evaluator.hs, dist\build\Clash\GHC\Evaluator.o )
[ 4 of 13] Compiling Clash.GHC.LoadInterfaceFiles ( src-ghc\Clash\GHC\LoadInterfaceFiles.hs, dist\build\Clash\GHC\LoadInterfaceFiles.o )
[ 5 of 13] Compiling Clash.GHC.LoadModules ( src-ghc\Clash\GHC\LoadModules.hs, dist\build\Clash\GHC\LoadModules.o )
[ 6 of 13] Compiling Clash.GHC.GenerateBindings ( src-ghc\Clash\GHC\GenerateBindings.hs, dist\build\Clash\GHC\GenerateBindings.o )
[ 7 of 13] Compiling Clash.GHC.NetlistTypes ( src-ghc\Clash\GHC\NetlistTypes.hs, dist\build\Clash\GHC\NetlistTypes.o )
[ 8 of 13] Compiling Clash.GHCi.UI.Info ( src-bin-841\Clash\GHCi\UI\Info.hs, dist\build\Clash\GHCi\UI\Info.o )
[ 9 of 13] Compiling Clash.GHCi.UI.Monad ( src-bin-841\Clash\GHCi\UI\Monad.hs, dist\build\Clash\GHCi\UI\Monad.o )
[10 of 13] Compiling Clash.GHCi.UI.Tags ( src-bin-841\Clash\GHCi\UI\Tags.hs, dist\build\Clash\GHCi\UI\Tags.o )
[11 of 13] Compiling Paths_clash_ghc  ( dist\build\autogen\Paths_clash_ghc.hs, dist\build\Paths_clash_ghc.o )
[12 of 13] Compiling Clash.GHCi.UI    ( src-bin-841\Clash\GHCi\UI.hs, dist\build\Clash\GHCi\UI.o )
[13 of 13] Compiling Clash.Main       ( src-bin-841\Clash\Main.hs, dist\build\Clash\Main.o )
Preprocessing executable 'clash' for clash-ghc-0.99.1..
Building executable 'clash' for clash-ghc-0.99.1..
[1 of 1] Compiling Main             ( src-ghc\Batch.hs, dist\build\clash\clash-tmp\Main.o )

src-ghc\Batch.hs:8:8: error:
    Could not find module ‘Prelude’
    Perhaps you haven't installed the "dyn" libraries for package ‘base-4.11.1.0’?
    Use -v to see a list of the files searched for.
  |
8 | module Main
  |        ^^^^

src-ghc\Batch.hs:12:1: error:
    Could not find module ‘System.Environment’
    Perhaps you haven't installed the "dyn" libraries for package ‘base-4.11.1.0’?
    Use -v to see a list of the files searched for.
   |
12 | import           System.Environment ( getArgs )
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

src-ghc\Batch.hs:13:1: error:
    Could not find module ‘Clash.Main’
    Perhaps you haven't installed the "dyn" libraries for package ‘clash-ghc-0.99.1’?
    Use -v to see a list of the files searched for.
   |
13 | import           Clash.Main         ( defaultMain )
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cabal: Leaving directory 'C:\Users\lenovo\AppData\Local\Temp\cabal-tmp-5024\clash-ghc-0.99.1'
```

chromezh, 2018-05-16:
I found a possible solution on https://stackoverflow.com/questions/7981350/cabal-rebuild-all-with-enable-shared-flag/7981580#7981580.

But after I tried, it seems that there remains the same problem.

Maybe I should install GHC in a different way?

```
> cabal install clash-ghc --enable-documentation --reinstall --enable-shared
Resolving dependencies...
Configuring clash-ghc-0.99.1...
Building clash-ghc-0.99.1...
Failed to install clash-ghc-0.99.1
Build log ( C:\Users\lenovo\AppData\Roaming\cabal\logs\ghc-8.4.2\clash-ghc-0.99.1-4esULADXYMUH3EBeTfhYob.log ):
Preprocessing library for clash-ghc-0.99.1..
Building library for clash-ghc-0.99.1..
[ 1 of 13] Compiling Clash.GHC.ClashFlags ( src-ghc\Clash\GHC\ClashFlags.hs, dist\build\Clash\GHC\ClashFlags.o )
[ 2 of 13] Compiling Clash.GHC.GHC2Core ( src-ghc\Clash\GHC\GHC2Core.hs, dist\build\Clash\GHC\GHC2Core.o )
[ 3 of 13] Compiling Clash.GHC.Evaluator ( src-ghc\Clash\GHC\Evaluator.hs, dist\build\Clash\GHC\Evaluator.o )
[ 4 of 13] Compiling Clash.GHC.LoadInterfaceFiles ( src-ghc\Clash\GHC\LoadInterfaceFiles.hs, dist\build\Clash\GHC\LoadInterfaceFiles.o )
[ 5 of 13] Compiling Clash.GHC.LoadModules ( src-ghc\Clash\GHC\LoadModules.hs, dist\build\Clash\GHC\LoadModules.o )
[ 6 of 13] Compiling Clash.GHC.GenerateBindings ( src-ghc\Clash\GHC\GenerateBindings.hs, dist\build\Clash\GHC\GenerateBindings.o )
[ 7 of 13] Compiling Clash.GHC.NetlistTypes ( src-ghc\Clash\GHC\NetlistTypes.hs, dist\build\Clash\GHC\NetlistTypes.o )
[ 8 of 13] Compiling Clash.GHCi.UI.Info ( src-bin-841\Clash\GHCi\UI\Info.hs, dist\build\Clash\GHCi\UI\Info.o )
[ 9 of 13] Compiling Clash.GHCi.UI.Monad ( src-bin-841\Clash\GHCi\UI\Monad.hs, dist\build\Clash\GHCi\UI\Monad.o )
[10 of 13] Compiling Clash.GHCi.UI.Tags ( src-bin-841\Clash\GHCi\UI\Tags.hs, dist\build\Clash\GHCi\UI\Tags.o )
[11 of 13] Compiling Paths_clash_ghc  ( dist\build\autogen\Paths_clash_ghc.hs, dist\build\Paths_clash_ghc.o )
[12 of 13] Compiling Clash.GHCi.UI    ( src-bin-841\Clash\GHCi\UI.hs, dist\build\Clash\GHCi\UI.o )
[13 of 13] Compiling Clash.Main       ( src-bin-841\Clash\Main.hs, dist\build\Clash\Main.o )
[ 1 of 13] Compiling Clash.GHC.ClashFlags ( src-ghc\Clash\GHC\ClashFlags.hs, dist\build\Clash\GHC\ClashFlags.dyn_o )

src-ghc\Clash\GHC\ClashFlags.hs:11:8: error:
    Could not find module ‘Prelude’
    Perhaps you haven't installed the "dyn" libraries for package ‘base-4.11.1.0’?
    Use -v to see a list of the files searched for.
   |
11 | module Clash.GHC.ClashFlags
   |        ^^^^^^^^^^^^^^^^^^^^

src-ghc\Clash\GHC\ClashFlags.hs:16:1: error:
    Could not find module ‘CmdLineParser’
    Perhaps you haven't installed the "dyn" libraries for package ‘ghc-8.4.2’?
    Use -v to see a list of the files searched for.
   |
16 | import CmdLineParser
   | ^^^^^^^^^^^^^^^^^^^^

src-ghc\Clash\GHC\ClashFlags.hs:17:1: error:
    Could not find module ‘Panic’
    Perhaps you haven't installed the "dyn" libraries for package ‘ghc-8.4.2’?
    Use -v to see a list of the files searched for.
   |
17 | import Panic
   | ^^^^^^^^^^^^

src-ghc\Clash\GHC\ClashFlags.hs:18:1: error:
    Could not find module ‘SrcLoc’
    Perhaps you haven't installed the "dyn" libraries for package ‘ghc-8.4.2’?
    Use -v to see a list of the files searched for.
   |
18 | import SrcLoc
   | ^^^^^^^^^^^^^

src-ghc\Clash\GHC\ClashFlags.hs:20:1: error:
    Could not find module ‘Data.IORef’
    Perhaps you haven't installed the "dyn" libraries for package ‘base-4.11.1.0’?
    Use -v to see a list of the files searched for.
   |
20 | import Data.IORef
   | ^^^^^^^^^^^^^^^^^

src-ghc\Clash\GHC\ClashFlags.hs:21:1: error:
    Could not find module ‘Control.Monad’
    Perhaps you haven't installed the "dyn" libraries for package ‘base-4.11.1.0’?
    Use -v to see a list of the files searched for.
   |
21 | import Control.Monad
   | ^^^^^^^^^^^^^^^^^^^^

src-ghc\Clash\GHC\ClashFlags.hs:22:1: error:
    Could not find module ‘Clash.Driver.Types’
    Perhaps you haven't installed the "dyn" libraries for package ‘clash-lib-0.99.1’?
    Use -v to see a list of the files searched for.
   |
22 | import Clash.Driver.Types
   | ^^^^^^^^^^^^^^^^^^^^^^^^^

src-ghc\Clash\GHC\ClashFlags.hs:23:1: error:
    Could not find module ‘Clash.Netlist.BlackBox.Types’
    Perhaps you haven't installed the "dyn" libraries for package ‘clash-lib-0.99.1’?
    Use -v to see a list of the files searched for.
   |
23 | import Clash.Netlist.BlackBox.Types (HdlSyn (..))
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

src-ghc\Clash\GHC\ClashFlags.hs:24:1: error:
    Could not find module ‘Text.Read’
    Perhaps you haven't installed the "dyn" libraries for package ‘base-4.11.1.0’?
    Use -v to see a list of the files searched for.
   |
24 | import Text.Read (readMaybe)
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cabal: Leaving directory 'C:\Users\lenovo\AppData\Local\Temp\cabal-tmp-1192\clash-ghc-0.99.1'
cabal.exe: Error: some packages failed to install:
clash-ghc-0.99.1-4esULADXYMUH3EBeTfhYob failed during the building phase. The
exception was:
ExitFailure 1
```

leonschoorl, 2018-05-18:
@chromezh: the problem is we're forcing dynamic linking of the clash binaries, which isn't supported by GHC on Windows.

To workaround this you could try the following:
```
cabal unpack clash-ghc
```
This creates a directory `clash-ghc-0.99.1`.
In there is the file `clash-ghc.cabal`, edit it and replace the 2 lines containing:
```
  GHC-Options:        -Wall -dynamic
```
with:
```
  GHC-Options:        -Wall
```

Then in that directory you can do:
```
cabal install --enable-documentation --dry-run
```

chromezh, 2018-05-18:
@leonschoorl 

It works. Thanks!

christiaanb, 2018-07-26:
The latest version no longer forces dynamic linking, so this should be fixed now.

closed 2018-07-26 by christiaanb