clash-lang/clash-compiler #316: SystemVerilog code generation bugs

bdmagnuson, 2018-05-18:
Design Compiler is a stickler for the standard so there are two problems with things like the below.

```
    function array_of_10_array_of_10_signed_15 array_of_10_array_of_10_signed_15_cons(dct2d_types::array_of_10_signed_15 x,logic signed [0:9][14:0] xs [0:8]);
     array_of_10_array_of_10_signed_15_cons[0] = {dct2d_types::array_of_10_signed_15_to_lv(x)};
     array_of_10_array_of_10_signed_15_cons[1:9] = xs;
   endfunction
```
First. the synthesizable SystemVerilog subset requires that each task and function in $unit (formally $root), in a package declaration, or in an interface declaration be declared automatic.

Second is that the first argument of the function references the package dct2d within the body of the package dct2d itself.  The package resolution portion should simply be left off here.

christiaanb, 2018-05-25:
Fair enough, I've only simulated the generated SV in modelsim (which didn't complain). Will fix.

christiaanb, 2019-08-14:
Fixed in #708

closed 2019-08-14 by christiaanb