clash-lang/clash-compiler #317: List entities and test benches from the command line / ghci

dented42, 2018-05-18:
It would be really helpful if there was some sort of `--listentities` and `--listtestbenches` flag that could be passed to the `clash` executable in place of `--vhdl` or `--verilog` that prints out a list of the entity names (and their testbenches).

Currently the build system I've written doesn't have any easy way of determining the names and locations of the HDL output, being able to ask clash for their names that I could derive the locations from.

dented42, 2018-05-18:
Or perhaps have them just print out the `Synthesize` and `TestBench` values raw to let tools parse them as they will?

martijnbastiaan, 2021-03-19:
Clash now outputs a `clash-manifest.json` file. Over time we will add more information to it (see issues [labeled "tooling integration"](https://github.com/clash-lang/clash-compiler/issues?q=is%3Aissue+is%3Aopen+label%3A%22tooling+integration%22)), but it should already be pretty useful.

Example json:

```json
{
    "components": [
        "addBound"
    ],
    "flags": [
        20,
        20,
        false
    ],
    "hash": 7931337818705174655,
    "dependencies": {
        "transitive": [
            "AddBound.addBoundLib"
        ]
    },
    "version": "unstable",
    "files": [
        {
            "name": "addBound.sdc",
            "sha256": "c564d4e38bc88a702a2909c47b587a2dfd8a814d68606d0691a5c0535a3fc8a5"
        },
        {
            "name": "addBound_types.vhdl",
            "sha256": "5bb7656c0a8c93e2fa42e64505606b4827b623a6e43659315bf735c1a7c6290a"
        },
        {
            "name": "addBound.vhdl",
            "sha256": "c4aa9528852683531e7ea72d26b4503c157bea36c2d4613c2bc71e95b79b1bb6"
        }
    ],
    "top_component": {
        "ports_flat": {
            "in": [
                {
                    "domain": "System",
                    "width": 1,
                    "name": "clock",
                    "is_clock": true,
                    "type_name": "clock.addBound_types.clk_System"
                },
                {
                    "domain": "System",
                    "width": 1,
                    "name": "reset",
                    "is_clock": false,
                    "type_name": "reset.addBound_types.rst_System"
                },
                {
                    "domain": "System",
                    "width": 1,
                    "name": "enable",
                    "is_clock": false,
                    "type_name": "enable.addBound_types.en_System"
                },
                {
                    "width": 33,
                    "name": "i",
                    "is_clock": false,
                    "type_name": "i.addBound_types.Maybe"
                }
            ],
            "out": [
                {
                    "width": 17,
                    "name": "o",
                    "is_clock": false,
                    "type_name": "o.addBound_types.Maybe"
                }
            ]
        },
        "name": "addBound"
    },
    "domains": {
        "System": {
            "period": 10000,
            "init_behavior": "Defined",
            "reset_kind": "Asynchronous",
            "active_edge": "Rising",
            "reset_polarity": "ActiveHigh"
        },
        "XilinxSystem": {
            "period": 10000,
            "init_behavior": "Defined",
            "reset_kind": "Synchronous",
            "active_edge": "Rising",
            "reset_polarity": "ActiveHigh"
        },
        "IntelSystem": {
            "period": 10000,
            "init_behavior": "Defined",
            "reset_kind": "Asynchronous",
            "active_edge": "Rising",
            "reset_polarity": "ActiveHigh"
        }
    }
}
```

closed 2021-03-19 by martijnbastiaan