clash-lang/clash-compiler #319: Issues with generated VHDL in the Tutorial

Nrmize, 2018-05-25:
Hi all,

I recently started using CLASH and got it installed on my machine with the following newest version.
>CLaSHi, version 0.99.1 (using clash-lib, version 0.99.1):

I was following along in the tutorial when I got down to the section labelled **Synthesize annotations**. The generated VHDL code that I am receiving from my CLASH compiler looks totally different than what is shown in the tutorial. 

**Here is my input Blinker.hs**
`module Blinker where

import Clash.Prelude
import Clash.Xilinx.ClockGen

type Dom50 = Dom "System" 20000

{-# ANN topEntity
  (Synthesize
    { t_name   = "blinker"
    , t_inputs = [ PortName "CLOCK_50"
                 , PortName "KEY0"
                 , PortName "KEY1"
                 ]
    , t_output = PortName "LED"
    }) #-}

topEntity :: Clock Dom50 Source -> Reset Dom50 Asynchronous -> Signal Dom50 Bit -> Signal Dom50 (BitVector 8)
topEntity clk rst = exposeClockReset (\key1 -> let key1R = isRising 1 key1 in  mealy blinkerT (1,False,0) key1R) pllOut rstSync
  where
    (pllOut,pllStable) = clockWizard @Dom50 (SSymbol @"clockWizardDiv2") clk rst
    rstSync            = resetSynchronizer pllOut (unsafeToAsyncReset pllStable)

blinkerT (leds,mode,cntr) key1R = ((leds',mode',cntr'),leds)
  where
    -- clock frequency = 50e6  (50 MHz)
    -- led update rate = 333e-3 (every 333ms)
    cnt_max = 16650000 -- 50e6 * 333e-3

    cntr' | cntr == cnt_max = 0
          | otherwise       = cntr + 1

    mode' | key1R     = not mode
          | otherwise = mode

    leds' | cntr == 0 = if mode then complement leds
                                else rotateL leds 1
          | otherwise = leds`

**Here is my generated Blinker.vhdl:**
`-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.blinker_types.all;

entity blinker is
  port(-- clock
       CLOCK_50 : in std_logic;
       -- asynchronous reset: active high
       KEY0     : in std_logic;
       KEY1     : in std_logic;
       LED      : out std_logic_vector(7 downto 0));
end;

architecture structural of blinker is
  signal \#app_arg\           : boolean;
  signal result               : std_logic;
  signal \#$d(%,%)_app_arg\   : boolean;
  signal \pllOut1\            : std_logic;
  signal \#$d(%,%)_app_arg_0\ : boolean;
  signal rst                  : std_logic;
  signal ds                   : blinker_types.tup2;
  signal \pllStable\          : boolean;
begin
  blinker_blinkert_sblinkert_led : entity blinker_blinkert_sblinkert
    port map
      ( leds => LED
      , clk  => \pllOut1\
      , rst  => result
      , i1   => \#app_arg\ );

  blinker_isrising_app_arg : entity blinker_isrising
    port map
      ( result => \#app_arg\
      , clk    => \pllOut1\
      , rst    => result
      , s      => KEY1 );

  result <= '1' when \#$d(%,%)_app_arg\ else '0';

  -- register begin 
  blinker_register : process(\pllOut1\,rst)
  begin
    if rst = '1' then
      \#$d(%,%)_app_arg\ <= true
      -- pragma translate_off
      after 1 ps
      -- pragma translate_on
      ;
    elsif rising_edge(\pllOut1\) then
      \#$d(%,%)_app_arg\ <= \#$d(%,%)_app_arg_0\
      -- pragma translate_off
      after 1 ps
      -- pragma translate_on
      ;
    end if;
  end process;
  -- register end

  \pllOut1\ <= ds.tup2_sel0;

  -- register begin 
  blinker_register_0 : process(\pllOut1\,rst)
  begin
    if rst = '1' then
      \#$d(%,%)_app_arg_0\ <= true
      -- pragma translate_off
      after 1 ps
      -- pragma translate_on
      ;
    elsif rising_edge(\pllOut1\) then
      \#$d(%,%)_app_arg_0\ <= false
      -- pragma translate_off
      after 1 ps
      -- pragma translate_on
      ;
    end if;
  end process;
  -- register end

  rst <= '1' when \pllStable\ else '0';

  -- clockWizard begin
  clockwizard : block
    signal pllout  : std_logic;
    signal locked  : std_logic;
    signal plllock : boolean;

    component clockWizardDiv2
      port (CLK_IN1  : in std_logic;
            RESET    : in std_logic;
            CLK_OUT1 : out std_logic;
            LOCKED   : out std_logic);
    end component;
  begin
    clockwizard_inst : component clockWizardDiv2 port map (CLOCK_50,KEY0,pllout,locked);
    plllock <= true when locked = '1' else false;
    ds <= (pllout,plllock);
  end block;
  -- clockWizard end

  \pllStable\ <= ds.tup2_sel1;
end;`

**Here is what is shown in the tutorial as the expected output:**
`-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.blinker_types.all;

entity blinker is
  port(-- clock
       CLOCK_50 : in std_logic;
       -- asynchronous reset: active high
       KEY0     : in std_logic;
       KEY1     : in std_logic_vector(0 downto 0);
       LED      : out std_logic_vector(7 downto 0));
end;

architecture structural of blinker is
begin
  blinker_topentity_inst : entity blinker_topentity
    port map
      (clk    => CLOCK_50
      ,rst    => KEY0
      ,key1   => KEY1
      ,result => LED);
end;`

As you can see, several wire names in the VHDL code were not named properly as far as I can tell (e.g. signal \#$d(%,%)_app_arg\ ). I have also have my generated VHDL for the FIR.hs from the Usage section of the main website. The generated FIR VHDL also exhibits improper naming conventions for the internal wires.

**FIR_topentity.vhdl**
`-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.fir_types.all;

entity fir_topentity is
  port(-- clock
       clk    : in std_logic;
       -- asynchronous reset: active high
       rst    : in std_logic;
       eta1   : in signed(15 downto 0);
       result : out signed(15 downto 0));
end;

architecture structural of fir_topentity is
  signal \#app_arg\ : fir_types.array_of_signed_16(0 to 3);
  signal bs         : fir_types.array_of_signed_16(0 to 3);
  signal \#vec1\    : fir_types.array_of_signed_16(0 to 3);
begin
  \#vec1\ <= fir_types.array_of_signed_16'( to_signed(2,16)
                                          , to_signed(3,16)
                                          , to_signed(-2,16)
                                          , to_signed(8,16) );

  -- zipWith begin
  zipwith : for i in \#app_arg\'range generate
  begin
    fir_fsaturatingnumsigned_csatmult_0 : entity fir_fsaturatingnumsigned_csatmult
      port map
        ( \#case_alt\ => \#app_arg\(i)
        , a           => \#vec1\(i)
        , b           => bs(i) );
  end generate;
  -- zipWith end

  fir_window_bs : entity fir_window
    port map
      ( result => bs
      , clk    => clk
      , rst    => rst
      , x      => eta1 );

  -- fold begin
  fold : block
    -- given a level and a depth, calculate the corresponding index into the
    -- intermediate array
    function depth2index (levels,depth : in natural) return natural is
    begin
      return (2 ** levels - 2 ** depth);
    end function;

    signal intermediate : fir_types.array_of_signed_16(0 to (2*4)-2);
    constant levels : natural := natural (ceil (log2 (real (4))));
  begin
    -- put input array into the first half of the intermediate array
    intermediate(0 to 4-1) <= \#app_arg\;

    -- Create the tree of instantiated components
    make_tree : if levels /= 0 generate
      tree_depth : for d in levels-1 downto 0 generate
        tree_depth_loop : for i_0 in 0 to (natural(2**d) - 1) generate
          fir_fsaturatingnumsigned_csatplus_1 : entity fir_fsaturatingnumsigned_csatplus
      port map
        ( \#case_alt\ => intermediate(depth2index(levels+1,d+1)+i_0)
        , a           => intermediate(depth2index(levels+1,d+2)+(2*i_0))
        , b           => intermediate(depth2index(levels+1,d+2)+(2*i_0)+1) );
        end generate;
      end generate;
    end generate;

    -- The last element of the intermediate array holds the result
    result <= intermediate((2*4)-2);
  end block;
  -- fold end
end;`

christiaanb, 2018-05-25:
Those names are proper extended VHDL identifiers, hence the surrounding `\ ... \`

Nrmize, 2018-05-25:
Thanks for the quick reply,

Then the generated output that I received is correct? I am a bit confused as to why these two VHDL outputs differ as much as they do.

**Tutorial output**
```
architecture structural of blinker is
begin
  blinker_topentity_inst : entity blinker_topentity
    port map
      (clk    => CLOCK_50
      ,rst    => KEY0
      ,key1   => KEY1
      ,result => LED);
end;
```

**My Actual output**
```
architecture structural of blinker is
  signal \#app_arg\           : boolean;
  signal result               : std_logic;
  signal \#$d(%,%)_app_arg\   : boolean;
  signal \pllOut1\            : std_logic;
  signal \#$d(%,%)_app_arg_0\ : boolean;
  signal rst                  : std_logic;
  signal ds                   : blinker_types.tup2;
  signal \pllStable\          : boolean;
begin
  blinker_blinkert_sblinkert_led : entity blinker_blinkert_sblinkert
    port map
      ( leds => LED
      , clk  => \pllOut1\
      , rst  => result
      , i1   => \#app_arg\ );

  blinker_isrising_app_arg : entity blinker_isrising
    port map
      ( result => \#app_arg\
      , clk    => \pllOut1\
      , rst    => result
      , s      => KEY1 );
```

christiaanb, 2018-05-25:
My bad, I didn't update the tutorial, I'll fix it in the next release

ameetgohil, 2018-06-26:
The current output doesn't seem like a valid syntax. Is there a step after ":verilog" or ":vhdl" to replace the \#\ to valid characters?

christiaanb, 2018-06-26:
Extended identifiers have been in the VHDL standard since VHDL-93, so those characters are valid. The top comment in the VHDL generated by the Clash compiler has always clearly stated that the generated  code adheres to the VHDL-93 standard.

ameetgohil, 2018-06-26:
Ah, I was unaware. Thanks! How do I go about compiling the verilog output with verilator? How does the .manifest file come into play?

ameetgohil, 2018-06-26:
Nevermind. I figured out compile with verilator

leonschoorl, 2018-06-26:
The .manifest file is used by clash itself to avoid unnecessary recompilation of (parts of) your design.
It's not meant for other tools.

christiaanb, 2018-07-26:
Tutorial has been fixed

closed 2018-07-26 by christiaanb