clash-lang/clash-compiler #320: Not able to create primitive

adamwalker, 2018-06-03:
I need an asymmetric width block ram in my design. The one in the Prelude is symmetric only so I decided to create a primitive. 

However, most of the time, my primitive is ignored. Where I would expect the primtive to be placed all I see is: 

```
  assign result = {32 {1'bx}};
```
The files to reproduce this are here: https://gist.github.com/adamwalker/cc30dc3f2b4d0ad6a4f37f66ab4b41d1.

If I replace the topEntity definition with:
```
topEntity = asymBlockRam# 
```
i.e. I remove the clk argument, then the ram is generated correctly. This is the only case where I have seen it generate the correct verilog. I am using the latest master (e1bc893).

leonschoorl, 2018-06-04:
Try changing your definition of asymBlockRam# to:
```haskell
asymBlockRam# clk = undefined
```

adamwalker, 2018-06-06:
Yeah, that works. As long as I list all the arguments to asymBlockRam it works fine. 

The problem is that this is unexpected. I think that when most people first create a primitive they will not bother with a Haskell implementation and simply leave it undefined without listing all the arguments, as I did. Clash will appear to ignore the primitive and they will be quite confused. 

On a related note, I now have an asymmetric block ram primitive that appears to work. If you would like me to incorporate that into the Prelude, I'd be happy to.

leonschoorl, 2018-06-07:
Yes, you are right, you should always list all the arguments to your custom primitives (if they're implemented with `error` or `undefined`).
And this is unexpected and annoying.

**TDLR**: When defining your own primitive and leaving the Haskell implementation undefined, list all the arguments of your primitive on the left side of the `=`
Like this: `myPrimitive _ _ = undefined`

The problem is we would like GHC(the frontend of Clash) to treat primitives as blackboxes, but when we do:
```haskell
myPrimitive :: Bool -> Bool -> Bool
myPrimitive = undefined
{-# NOINLINE myPrimitive #-}

topEntity a b = myPrimitive a b
```

We would expect GHC to translate `topEntity` to:
```haskell
topEntity = \a b -> myPrimitive a b
```
But GHC looks into `myPrimitive` and sees that we're applying `a` and `b` as the arguments to `undefined`.
 knows that that will never produce a (meaningful) result and produces:
```haskell
topEntity = \a b -> case myPrimitive a b of
  -- EmptyCase
```
Even if Clash replaces `myPrimitive` with a HDL template that is defined, `topEntity` is now undefined.

It seems hard to fix this properly without changing GHC.
But I'm looking into having Clash produce a warning/error when you're defining a (undefined) primitive without listing all of it's arguments.
That should make it more clear to users what's happening.

christiaanb, 2018-06-25:
I should update the tutorial: to make sure GHC doesn't mess about with your primitives, ensure that the function "implementing" the primitive adheres to the following rules:

1. All its arguments must have a demand
2. Reducing the result to WHNF must _not_ result in an error.

So in the case of your example, this is would be the preferred implementation:

```haskell
{-# LANGUAGE BangPatterns #-}

asymBlockRam#
  :: HasCallStack    --       ARG[0]
  => Clock dom gated -- clk,  ARG[1]
  -> Vec n a         -- init, ARG[2]
  -> Signal dom Int  -- rd,   ARG[3]
  -> Signal dom Bool -- wren, ARG[4]
  -> Signal dom Int  -- wr,   ARG[5]
  -> Signal dom a    -- din,  ARG[6]
  -> Signal dom (Vec m a)
asymBlockRam# !_clk !_init !_rd !_wren !_wr !_din = pure undefined
```

where the bang patterns ensure your arguments have a demand, and saying `pure undefined` instead of `undefined` ensure that the `Signal` can reduce to WHNF without throwing an error.

christiaanb, 2018-06-25:
So a proper solution would be to write a core-to-core plugin for GHC that ensures that functions that you annotate as a primitive get a specific demand/strictness signature so that GHC doesnt mess about with it later.

A faster "solution" would be that Clash checks the demand/strictness signature of primitives and throw an error when the demand signature is indicative of GHC later messing about with the primitive.

christiaanb, 2018-12-18:
Flagging this as milestone 1.0 since we've already put some work into the checks

commit c23cb846 (2019-08-14): Only report NoDemand for arguments used in BlackBox

Previously, we received warnings when any of the arguments of
the Haskell implementation of a primitive had a `Absent`
demand signature; even if those arguments weren't used in the
BlackBox for the primitive either.

Now we only get a warning when an argument that's used in the
BlackBox has an `Absent` demand signature. The reason why this
situation is problematic is that expressions occurring in the
argument position of applications of this primitive are
sometimes replaced by `absentArg` by GHC; because the Haskell
implementation wasn't usi [... cut]
commit 59239e5d (2019-08-14): Only report NoDemand for arguments used in BlackBox

Previously, we received warnings when any of the arguments of
the Haskell implementation of a primitive had a `Absent`
demand signature; even if those arguments weren't used in the
BlackBox for the primitive either.

Now we only get a warning when an argument that's used in the
BlackBox has an `Absent` demand signature. The reason why this
situation is problematic is that expressions occurring in the
argument position of applications of this primitive are
sometimes replaced by `absentArg` by GHC; because the Haskell
implementation wasn't usi [... cut]
commit abe10346 (2019-08-14): Only report NoDemand for arguments used in BlackBox

Previously, we received warnings when any of the arguments of
the Haskell implementation of a primitive had a `Absent`
demand signature; even if those arguments weren't used in the
BlackBox for the primitive either.

Now we only get a warning when an argument that's used in the
BlackBox has an `Absent` demand signature. The reason why this
situation is problematic is that expressions occurring in the
argument position of applications of this primitive are
sometimes replaced by `absentArg` by GHC; because the Haskell
implementation wasn't usi [... cut]
commit dac4457a (2019-08-14): Only report NoDemand for arguments used in BlackBox

Previously, we received warnings when any of the arguments of
the Haskell implementation of a primitive had a `Absent`
demand signature; even if those arguments weren't used in the
BlackBox for the primitive either.

Now we only get a warning when an argument that's used in the
BlackBox has an `Absent` demand signature. The reason why this
situation is problematic is that expressions occurring in the
argument position of applications of this primitive are
sometimes replaced by `absentArg` by GHC; because the Haskell
implementation wasn't usi [... cut]
closed 2019-08-14 by christiaanb