clash-lang/clash-compiler #323: Clash generates incorrect verilog with nested maps

adamwalker, 2018-06-24:
I believe that clash-ghc-0.99.1 generates incorrect verilog for this file: https://gist.github.com/adamwalker/7d0365a1df54c2d99b2a871d8e6852f4

Specifically, when the "inRange" module is instantiated in the top level module, the input signal named "current"  is connected to the wrong signal. In the generated code it is connected to map_in_1, which is also the wrong width for the "current" input port. I believe it should be connected to map_in_2, which is not used anywhere.

christiaanb, 2018-06-25:
Interesting, this seems to be a substitution capture problem in the template handling code. I'm on it.

christiaanb, 2018-06-25:
Turned out it was just a scoping issue, fix wasn't too difficult.

commit ff48c61c (2018-06-25): Use correct arguments in nested primitives

Without scoping knowledge, the blackbox template instantiation
logic always picked the closest argument.

Fixes #323
closed 2018-06-25 by christiaanb