clash-lang/clash-compiler #328: Synthesize annotations are ignored

adamwalker, 2018-07-11:
If I generate Verilog using Clash 0.99.2 for the Clash code below, the Synthesize annotation seems to be ignored. 

```
import Clash.Prelude

{-# ANN func
  (Synthesize
    { t_name   = "test"
    , t_inputs = [ PortName "clk"
                 , PortName "reset"
                 , PortName "x" ]
    , t_output = PortName "y"
    }) #-}
func 
    :: Clock System Source 
    -> Reset System Synchronous 
    -> Signal System Bool 
    -> Signal System Bool
func clk reset p = withClockReset clk reset $ register False p
```
I get the error:
```
<no location info>: error:
    Other error:
    No 'topEntity', nor function with a 'Synthesize' annotation found in root module: Main
```
Compilation succeeds if I rename the function to topEntity, but then the port and module names are wrong.

christiaanb, 2018-07-11:
Odd.. I can't replicate with `master`

adamwalker, 2018-07-11:
I rebuilt Clash from master and the problem went away. I had previously built Clash with ghc 8.4.2. I then upgraded to the latest Stackage LTS which uses 8.4.3 but I don't think I rebuilt Clash. Could that explain it?

christiaanb, 2018-07-12:
Possibly, I was building with GHC 8.4.1

adamwalker, 2018-07-15:
I rebuilt Clash with GHC 8.4.1, 8.4.2, and 8.4.3 and I can no longer reproduce this. This is super weird because I still have the terminal open and can see the error so I didn't imagine it. Anyway, since I can no longer reproduce this, and everything works great, it's probably safe to close the issue.

pacak, 2019-04-04:
Same problem with ghc 8.4.4 and 0.99.3. If name is not `topEntry` - compilation fails, if name is `topEntry` - anything inside annotation is ignored.

pacak, 2019-04-04:
It was achieved in clean  installation of ghc 8.4.4 (via ghcup), clash compiled with ` cabal new-install clash-ghc --enable-documentation --enable-executable-dynamic` (at least that's what I found in my command line history) then created cabal project with `build-depends:       base ^>=4.11.1.0, clash-prelude`.

Problem was fixed by installing clash-ghc with stack and using it as stack project. So something fishy is going on.

adamwalker, 2019-04-04:
I've seen this again a couple of times since I originally created the issue. Deleting the .stack-work directory and rebuilding seems to fix.

leonschoorl, 2019-04-04:
`ANN`otations are [serialized](https://downloads.haskell.org/~ghc/8.6.4/docs/html/libraries/ghc-boot-8.6.4/GHC-Serialized.html) using Data.Data and Data.Typeable.
And Typeable's TypeRep contains the unit id of the package the type comes from.

So for clash to recognize the annotations, your clash design has to be compiled against the exact same clash-prelude(with the same unit id) that clash compiler was compiled against.

If cabal/stack decides to rebuild clash-prelude for some reason (maybe because some dependency was updated or some configuration was changed), and it doesn't rebuild clash-ghc then clash won't recognize annotations anymore.

martijnbastiaan, 2019-12-20:
We're now recognizing this case and throwing an error. Thanks for the report!

closed 2019-12-20 by martijnbastiaan