clash-lang/clash-compiler #331: BlockRam initialisation on Verilog not executing properly.

anmolsahoo25, 2018-07-24:
Initialising a blockRam in Clash gives the following output - 

```
// blockRam begin                                                                                                                                            
  reg signed [31:0] RAM [0:128-1];                                                                                                                             
                                                                                                                                                               
  reg [4095:0] ram_init;                                                                                                                                       
  integer i;                                                                                                                                                   
  initial begin                                                                                                                                                
    ram_init = xs;                                                                                                                                             
    for (i=0; i < 128; i = i + 1) begin                                                                                                                        
      RAM[128-1-i] = ram_init[i*32+:32];                                                                                                                       
    end                                                                                                                                                        
  end                                                                                                                                                          
                                                                                                                                                               
  always @(posedge CLK) begin                                                                                                                                  
      ram_init <= xs;                                                                                                                                          
  end                                                                                                                                                          
                                                                                                                                                               
  always @(posedge CLK) begin : memElement_blockRam                                                                                                            
    if (\#app_arg ) begin                                                                                                                                      
      RAM[(wild_0)] <= y;                                                                                                                                      
    end                                                                                                                                                        
    RD_DATA <= RAM[(wild)];                                                                                                                                    
  end                                                                                                                                                          
  // blockRam end
```

But as can be seen in this part, 
```
initial begin                                                                                                                                                
    ram_init = xs;                                                                                                                                             
    for (i=0; i < 128; i = i + 1) begin                                                                                                                        
      RAM[128-1-i] = ram_init[i*32+:32];                                                                                                                       
    end                                                                                                                                                        
  end 
```
The initial value of wires is always z (High impedance), thus for Verilog Simulators, this does not work as subsequent cycles all have the value of RAM instantiated to z.

One possible solution is to add this code
```
always @(posedge clock) begin
    ram_init <= xs;
end
```

Could anyone comment on this? I am using iVerilog on my machine, simulation using vvp and plotting the signal output using GTKwave.

christiaanb, 2018-07-24:
Clash should actually error out when the initial value of a block ram is not a literal. It does that when generating VHDL or SystemVerilog, but not for Verilog.

ggreif, 2018-07-24:
@anmolsahoo25 you can probably generate the vector by `-XTemplateHaskell`, in which case your module will see a literal value.

anmolsahoo25, 2018-07-24:
@christiaanb @ggreif Thank you. That seems like a good solution for now, wasn't aware of Template Haskell. I'll try looking around if I can fix the Verilog Backend for this bug.

leonschoorl, 2018-07-24:
The bug is in the primitive definition of blockRam:
https://github.com/clash-lang/clash-compiler/blob/52deb67fb67cbae08837f2fc0a89be845b38a9b1/clash-lib/prims/verilog/Clash_Explicit_BlockRam.json#L21
it should assert that initial state(`ARG[2]`) is a literal:
```
  ~SYM[2] = ~LIT[2];
```
Just like the SystemVerilog version:
https://github.com/clash-lang/clash-compiler/blob/52deb67fb67cbae08837f2fc0a89be845b38a9b1/clash-lib/prims/systemverilog/Clash_Explicit_BlockRam.json#L18
And the VHDL version:
https://github.com/clash-lang/clash-compiler/blob/52deb67fb67cbae08837f2fc0a89be845b38a9b1/clash-lib/prims/vhdl/Clash_Explicit_BlockRam.json#L16

PR #333 (2018-11-02): The (initial) content for all blockRam's and rom's must be constant. Fix #331
Fixes #331
commit d9a30831 (2018-11-02): The (initial) content for all blockRam's and rom's must be constant

Fixes #331
closed 2018-11-02 by leonschoorl