clash-lang/clash-compiler #335: can't install 0.99.3 neither on linux nor on win10

postoroniy, 2018-07-28:
linux:
```
cabal install clash-ghc --enable-documentation --enable-executable-dynamic
Resolving dependencies...
Downloading clash-ghc-0.99.3...
Configuring clash-ghc-0.99.3...
Building clash-ghc-0.99.3...
Failed to install clash-ghc-0.99.3
Build log ( /home/slava/.cabal/logs/ghc-8.4.3/clash-ghc-0.99.3-CnJL8sQpIv12KwT8rGz6m0.log ):
cabal: Entering directory '/tmp/cabal-tmp-10111/clash-ghc-0.99.3'
Configuring clash-ghc-0.99.3...
Preprocessing library for clash-ghc-0.99.3..
Building library for clash-ghc-0.99.3..
[ 1 of 13] Compiling Clash.GHC.ClashFlags ( src-ghc/Clash/GHC/ClashFlags.hs, dist/build/Clash/GHC/ClashFlags.o )
[ 2 of 13] Compiling Clash.GHC.GHC2Core ( src-ghc/Clash/GHC/GHC2Core.hs, dist/build/Clash/GHC/GHC2Core.o )
[ 3 of 13] Compiling Clash.GHC.Evaluator ( src-ghc/Clash/GHC/Evaluator.hs, dist/build/Clash/GHC/Evaluator.o )

src-ghc/Clash/GHC/Evaluator.hs:609:27: error:
    Not in scope: data constructor ‘ByteArrayLiteral’
    |
609 |            Just (Literal (ByteArrayLiteral (Vector.Vector voff vlen ba1))) =
    |                           ^^^^^^^^^^^^^^^^

src-ghc/Clash/GHC/Evaluator.hs:630:27: error:
    Not in scope: data constructor ‘ByteArrayLiteral’
    |
630 |            Just (Literal (ByteArrayLiteral (Vector.Vector off len ba1))) =
    |                           ^^^^^^^^^^^^^^^^

src-ghc/Clash/GHC/Evaluator.hs:657:13: error:
    Not in scope: data constructor ‘ByteArrayLiteral’
    |
657 |     | [Lit (ByteArrayLiteral ba)] <- args
    |             ^^^^^^^^^^^^^^^^

src-ghc/Clash/GHC/Evaluator.hs:661:13: error:
    Not in scope: data constructor ‘ByteArrayLiteral’
    |
661 |     | [Lit (ByteArrayLiteral (Vector.Vector _ _ (ByteArray.ByteArray ba))),iV] <- args
    |             ^^^^^^^^^^^^^^^^

src-ghc/Clash/GHC/Evaluator.hs:676:27: error:
    Not in scope: data constructor ‘ByteArrayLiteral’
    |
676 |            Just (Literal (ByteArrayLiteral ba')) = IntMap.lookup (fromInteger ba) gh
    |                           ^^^^^^^^^^^^^^^^

src-ghc/Clash/GHC/Evaluator.hs:692:27: error:
    Not in scope: data constructor ‘ByteArrayLiteral’
    |
692 |            Just (Literal (ByteArrayLiteral (Vector.Vector 0 _ ba1)))
    |                           ^^^^^^^^^^^^^^^^

src-ghc/Clash/GHC/Evaluator.hs:716:27: error:
    Not in scope: data constructor ‘ByteArrayLiteral’
    |
716 |            Just (Literal (ByteArrayLiteral (Vector.Vector voff vlen ba1))) =
    |                           ^^^^^^^^^^^^^^^^

src-ghc/Clash/GHC/Evaluator.hs:728:13: error:
    Not in scope: data constructor ‘ByteArrayLiteral’
    |
728 |     | [Lit (ByteArrayLiteral (Vector.Vector _ _ (ByteArray.ByteArray src_ba)))
    |             ^^^^^^^^^^^^^^^^

src-ghc/Clash/GHC/Evaluator.hs:736:27: error:
    Not in scope: data constructor ‘ByteArrayLiteral’
    |
736 |            Just (Literal (ByteArrayLiteral (Vector.Vector voff vlen dst_ba))) =
    |                           ^^^^^^^^^^^^^^^^

src-ghc/Clash/GHC/Evaluator.hs:759:27: error:
    Not in scope: data constructor ‘ByteArrayLiteral’
    |
759 |            Just (Literal (ByteArrayLiteral (Vector.Vector _ _ ba1))) =
    |                           ^^^^^^^^^^^^^^^^
cabal: Leaving directory '/tmp/cabal-tmp-10111/clash-ghc-0.99.3'
cabal: Error: some packages failed to install:
clash-ghc-0.99.3-CnJL8sQpIv12KwT8rGz6m0 failed during the building phase. The
exception was:
ExitFailure 1
```
win10 asking for ghc 8.4.2 while 8.4.3 is installed and  failed on the same step 3

christiaanb, 2018-07-28:
Oops... I forgot to update the lower bounds on the `clash-lib` and `clash-prelude` dependencies. I've just fixed that, if you do:

```
cabal update
cabal install clash-prelude-0.99.3 clash-lib-0.99.3 clash-ghc-0.99.3
```

everything will hopefully work.

postoroniy, 2018-07-29:
linux looks fine now but 
on win10  I got this after installing clash.

>clashi
>CLaSHi, version 0.99.3 (using clash-lib, version 0.99.3):
>http://www.clash-lang.org/  :? for help
>clashi: addLibrarySearchPath: C:\Program Files\Haskell Platform\8.4.2\mingw\lib (Win32 error 3): The >system cannot find the path specified.

I don'y have 8.4.2 anymore on win10 I have only 8.4.3

leonschoorl, 2018-07-29:
I just tried it with a clean install of Windows 10, using Haskell Core platform 8.4.3, and clash-0.99.3 works fine there.

There is something weird going on with ghc installation.
You should check that there is no mentions of 8.4.2 in any of your environment variables, especially your PATH.
```
C:\Users\user>set
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\user\AppData\Roaming
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=WIN10-VM
ComSpec=C:\Windows\system32\cmd.exe
HOMEDRIVE=C:
HOMEPATH=\Users\user
LOCALAPPDATA=C:\Users\user\AppData\Local
LOGONSERVER=\\WIN10-VM
NUMBER_OF_PROCESSORS=2
OS=Windows_NT
Path=C:\Program Files\Haskell\bin;C:\Program Files\Haskell Platform\8.4.3\lib\extralibs\bin;C:\Program Files\Haskell Platform\8.4.3\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Haskell Platform\8.4.3\mingw\bin;C:\Users\user\AppData\Roaming\cabal\bin;C:\Users\user\AppData\Local\Microsoft\WindowsApps
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PROCESSOR_ARCHITECTURE=AMD64
...
```

Check that cabal, ghc and ghci are of the right version, and ghci is working:
```
C:\Users\user>ghc --version
The Glorious Glasgow Haskell Compilation System, version 8.4.3

C:\Users\user>cabal --version
cabal-install version 2.2.0.0
compiled using version 2.2.0.1 of the Cabal library

C:\Users\user>ghci
GHCi, version 8.4.3: http://www.haskell.org/ghc/  :? for help
Prelude> putStrLn  "Hello World!"
Hello World!
Prelude> :quit
Leaving GHCi.
```

Then you could try removing all cabal's previously installed stuff , by deleting the directories: `%APPDATA%\cabal` and `%APPDATA%\cabal`:

```
C:\Users\user\Desktop>rd /S %APPDATA%\ghc
C:\Users\user\AppData\Roaming\ghc, Are you sure (Y/N)? y

C:\Users\user\Desktop>rd /S %APPDATA%\cabal
C:\Users\user\AppData\Roaming\cabal, Are you sure (Y/N)? y
```

After that you should be able to reinstall clash:
```
C:\Users\user\Desktop>cabal update
Config file path source is default config file.
Config file C:\Users\user\AppData\Roaming\cabal\config not found.
Writing default configuration to C:\Users\user\AppData\Roaming\cabal\config
Downloading the latest package list from hackage.haskell.org

C:\Users\user\Desktop>cabal install clash-ghc --enable-documentation
Resolving dependencies...
Downloading StateVar-1.1.1.0...
Downloading base-compat-0.10.4...
...
Installed clash-ghc-0.99.3

C:\Users\user\Desktop>clashi
CLaSHi, version 0.99.3 (using clash-lib, version 0.99.3):
http://www.clash-lang.org/  :? for help
Clash.Prelude> 2 * 0x10 + 10 :: BitVector 6
10_1010
Clash.Prelude>
```

postoroniy, 2018-07-30:
@leonschoorl  
I didn't have any residual of 8.4.2  I checked 
But your trick to delete  

> C:\Users\user\Desktop>rd /S %APPDATA%\ghc
> C:\Users\user\Desktop>rd /S %APPDATA%\cabal
 worked for me!

Thanks!

postoroniy, 2019-02-12:
I had to reinstall on linux and got this (cabal 2.4, ghc-8.4.3)

Resolving dependencies...
Warning: The following packages are likely to be broken by the reinstalls:
clash-utils-0.1.0.0
Continuing even though the plan contains dangerous reinstalls.
Starting     clash-prelude-0.99.3
Building     clash-prelude-0.99.3
Failed to install clash-prelude-0.99.3
Build log ( /home/slava/.cabal/logs/ghc-8.4.3/clash-prelude-0.99.3-BY1UV2vaXFRIvFFiAZTLRS.log ):
cabal: Entering directory '/tmp/cabal-tmp-14436/clash-prelude-0.99.3'
Configuring clash-prelude-0.99.3...
Preprocessing library for clash-prelude-0.99.3..
Building library for clash-prelude-0.99.3..
[ 1 of 66] Compiling Clash.Annotations.Primitive ( src/Clash/Annotations/Primitive.hs, dist/build/Clash/Annotations/Primitive.o )
[ 2 of 66] Compiling Clash.Annotations.TopEntity ( src/Clash/Annotations/TopEntity.hs, dist/build/Clash/Annotations/TopEntity.o )
[ 3 of 66] Compiling Clash.Class.BitPack.Internal ( src/Clash/Class/BitPack/Internal.hs, dist/build/Clash/Class/BitPack/Internal.o )
[ 4 of 66] Compiling Clash.Class.Num  ( src/Clash/Class/Num.hs, dist/build/Clash/Class/Num.o )
[ 5 of 66] Compiling Clash.Class.Resize ( src/Clash/Class/Resize.hs, dist/build/Clash/Class/Resize.o )
<command line>: Could not find module ‘GHC.TypeLits.KnownNat.Solver’
There are files missing in the ‘ghc-typelits-knownnat-0.5.1’ package,
try running 'ghc-pkg check'.
Use -v to see a list of the files searched for.
cabal: Leaving directory '/tmp/cabal-tmp-14436/clash-prelude-0.99.3'
cabal: Error: some packages failed to install:
clash-ghc-0.99.3-1oItzzdqmph87N2473H5JW depends on clash-ghc-0.99.3 which
failed to install.
clash-lib-0.99.3-9neuJCCBI8RCWlAKOUmaMP depends on clash-lib-0.99.3 which
failed to install.
clash-prelude-0.99.3-BY1UV2vaXFRIvFFiAZTLRS failed during the building phase.
The exception was:
ExitFailure 1

leonschoorl, 2019-02-12:
> : Could not find module ‘GHC.TypeLits.KnownNat.Solver’
> There are files missing in the ‘ghc-typelits-knownnat-0.5.1’ package,
> try running 'ghc-pkg check'.

You could try reinstalling ghc-typelits-knownnat-0.5.1:
```
cabal install  ghc-typelits-knownnat-0.5.1 --force-reinstalls
```

But the easiest way is probably to delete all previous cabal/ghc stuff and start fresh:
```
rm -r ~/.ghc ~/.cabal
```

postoroniy, 2019-02-12:
OK..solved..
Thanks

closed 2019-02-12 by postoroniy