clash-lang/clash-compiler #349: Partially applied argument in HO-primitives not globally static expressions

christiaanb, 2018-09-11:
In things like:

`\x y z -> map (f (x,y)) z`

Clash will generate the following vhdl:

```vhdl
  -- map begin
  map_r : for i in z'range generate
  begin
    f_0 : entity f
      port map
        (result  => result(i)
        ,inp1 => (product_sel0 => x, product_sel1 => y)
        ,inp2 => z(i) );
  end generate;
  -- map end
```

This is because when it sees that the HO-argument of `map` is a function: https://github.com/clash-lang/clash-compiler/blob/1d02f8fb9f576b0d00225cc94c9f9ee2ac3c7ef5/clash-lib/src/Clash/Netlist/BlackBox.hs#L370 it does __not__ ensure that the partially applied arguments storted in `bbCtx` https://github.com/clash-lang/clash-compiler/blob/1d02f8fb9f576b0d00225cc94c9f9ee2ac3c7ef5/clash-lib/src/Clash/Netlist/BlackBox.hs#L326 are simple variables. 

This is what Clash normally does when it translates function application into component instantiation: https://github.com/clash-lang/clash-compiler/blob/1d02f8fb9f576b0d00225cc94c9f9ee2ac3c7ef5/clash-lib/src/Clash/Netlist.hs#L491

So what needs to happen is that the blackBoxContext `bbCtx` only contains simple variables, when the HO-argument is a function.

martijnbastiaan, 2019-01-28:
I've found a small example triggering the bug: 

```haskell
module Map where

import Clash.Prelude
import Clash.Explicit.Testbench

go
  :: SystemClockReset
  => ()
  -> Signal System Bool
  -> Signal System Bool
go _ = mealy update initialState
  where
    initialState :: (Bool, Index 5)
    initialState = (False, minBound)

    update (c, index) _ = ((c, index), c)

testInput
  :: SystemClockReset
  => Signal System (Vec 4 Bool)
  -> Signal System (Vec 4 Bool)
testInput =
  bundle . fmap (go ()) . unbundle

topEntity = testInput
```

martijnbastiaan, 2019-01-28:
@christiaanb To solve this bug I basically generalized `setSym` such that it can be used in a `NetlistMonad` and `State backend` context. Then, I modified `renderElem` to handle the declarations `setSym` might generate. While the changes make sense to me, I can't really judge if I've missed something. Could you take a quick look at https://github.com/clash-lang/clash-compiler/commit/5985933948a79f6c0cde55b13d5a93205180c70b ? (Don't mind the code style, I'll fix it in the morning.)

christiaanb, 2019-01-28:
@martijnbastiaan yeah, looks good to me

commit 009902c1 (2019-01-29): Allow blackboxes generated by `mkFunInput` to use `~VAR`. Fixes #349
commit 46a5d6d0 (2019-01-29): Allow blackboxes generated by `mkFunInput` to use `~VAR`. Fixes #349
closed 2019-01-29 by martijnbastiaan